ros2/rosbag2 #25: rosbag_write_integration_test is flaky

Martin-Idel-SI, 2018-08-24:
Currently, the rosbag_write_integration_test is flaky.

Valgrind and the debugger both suggest that cleanup of `shared_ptr<SerializedBagMessage>` does not work properly. However, the code never calls `reset` explicitly, nor is the memory in question released in any other way we can see.

Martin-Idel-SI, 2018-08-31:
We have heavily improved the tests. Sadly, the `rosbag2_write_integration_test` (and the second write `rosbag2_write_all_integration_test`) are still flaky:

Linux seems fine now, but we get the error `Expression: vector erase iterator outside range` following stack trace on Windows, which seems unrelated to our code:

```
 	msvcp140d.dll!00007ffecd72fc06()	Unknown
 	fastrtpsd-1.6.dll!std::vector<std::pair<eprosima::fastrtps::Publisher * __ptr64,eprosima::fastrtps::PublisherImpl * __ptr64>,std::allocator<std::pair<eprosima::fastrtps::Publisher * __ptr64,eprosima::fastrtps::PublisherImpl * __ptr64> > >::erase(std::_Vector_const_iterator<std::_Vector_val<std::_Simple_types<std::pair<eprosima::fastrtps::Publisher *,eprosima::fastrtps::PublisherImpl *> > > > _Where) Line 1475	C++
 	fastrtpsd-1.6.dll!eprosima::fastrtps::ParticipantImpl::removePublisher(eprosima::fastrtps::Publisher * pub) Line 84	C++
 	fastrtpsd-1.6.dll!eprosima::fastrtps::Domain::removePublisher(eprosima::fastrtps::Publisher * pub) Line 93	C++
 	rmw_fastrtps_shared_cpp.dll!rmw_fastrtps_shared_cpp::__rmw_destroy_publisher(const char * identifier, rmw_node_t * node, rmw_publisher_t * publisher) Line 67	C++
 	rmw_fastrtps_cpp.dll!rmw_destroy_publisher(rmw_node_t * node, rmw_publisher_t * publisher) Line 189	C++
 	rmw_implementation.dll!rmw_destroy_publisher(rmw_node_t * v2, rmw_publisher_t * v1) Line 245	C++
 	rcl.dll!rcl_publisher_fini(rcl_publisher_t * publisher, rcl_node_t * node) Line 210	C
 	rclcpp.dll!rclcpp::PublisherBase::~PublisherBase() Line 90	C++
 	rosbag2_write_integration_test.exe!rclcpp::Publisher<std_msgs::msg::String_<std::allocator<void> >,std::allocator<void> >::~Publisher<std_msgs::msg::String_<std::allocator<void> >,std::allocator<void> >() Line 181	C++
 	[External Code]	
 	rosbag2_write_integration_test.exe!RosBag2WriteIntegrationTestFixture::start_publishing::__l2::<lambda>() Line 101	C++
 	[External Code]	
 	[Async Call]	
 	[External Code]	
 	rosbag2_write_integration_test.exe!RosBag2WriteIntegrationTestFixture::start_publishing(std::shared_ptr<rosbag2_storage::SerializedBagMessage> message, std::basic_string<char,std::char_traits<char>,std::allocator<char> > topic_name, unsigned __int64 number_expected_messages) Line 90	C++
 	rosbag2_write_integration_test.exe!RosBag2WriteIntegrationTestFixture_published_messages_from_multiple_topics_are_recorded_Test::TestBody() Line 43	C++
 	rosbag2_write_integration_test.exe!testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test,void>(testing::Test * object, void(testing::Test::*)() method, const char * location) Line 2380	C++
 	rosbag2_write_integration_test.exe!testing::internal::HandleExceptionsInMethodIfSupported<testing::Test,void>(testing::Test * object, void(testing::Test::*)() method, const char * location) Line 2431	C++
 	rosbag2_write_integration_test.exe!testing::Test::Run() Line 2474	C++
 	rosbag2_write_integration_test.exe!testing::TestInfo::Run() Line 2649	C++
 	rosbag2_write_integration_test.exe!testing::TestCase::Run() Line 2764	C++rosbag2_write_integration_test.exe!RosBag2WriteIntegrationTestFixture::start_publishing(std::shared_ptr<rosbag2_storage::SerializedBagMessage> message, std::basic_string<char,std::char_traits<char>,std::allocator<char> > topic_name, unsigned __int64 number_expected_messages) Line 90	C++
 	rosbag2_write_integration_test.exe!RosBag2WriteIntegrationTestFixture_published_messages_from_multiple_topics_are_recorded_Test::TestBody() Line 43	C++
 	rosbag2_write_integration_test.exe!testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test,void>(testing::Test * object, void(testing::Test::*)() method, const char * location) Line 2380	C++
  	rosbag2_write_integration_test.exe!testing::internal::HandleExceptionsInMethodIfSupported<testing::Test,void(testing::Test * object, void(testing::Test::*)() method, const char * location) Line 2431	C++
  	rosbag2_write_integration_test.exe!testing::Test::Run() Line 2474	C++
  	rosbag2_write_integration_test.exe!testing::TestInfo::Run() Line 2649	C++
  	rosbag2_write_integration_test.exe!testing::TestCase::Run() Line 2764	C++
``` 	

This error is thrown upon finishing our publishing async call, which automatically destroys the publisher (as seen from the stacktrace).

Karsten1987, 2018-09-28:
fixed in #38

closed 2018-09-28 by Karsten1987