ros2/rosbag2 #35: rosbag2_record_all_integration_test is flaky

anhosi, 2018-09-18:
Currently the rosbag2_record_all_integration_test test is flaky when running with FastRTPS and fails with the following error:

```
C++ exception with description "could not create publisher:
create_publisher() could not create publisher, at <ROS2_WS>/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/rmw_publisher.cpp:134, at <ROS2_WS>/src/ros2/rcl/rcl/src/rcl/publisher.c:173" 
thrown in the test body.
PARTICIPANT Error Type : test_msgs::msg::dds_::Primitives_ Not Registered -> Function createPublisher
```

@Karsten1987  Are we doing something wrong in rosbag2 or is this a rmw/ dds problem?

Karsten1987, 2018-09-20:
I can't reproduce this one. However, I've run into a couple of segfaults when running the `record_integration_test`:

```
[ RUN      ] RosBag2RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded
Database name: tmp_test_dir.03M0vq/published_messages_from_multiple_topics_are_recorded.db3
[INFO] [rosbag2_storage]: Opened database 'tmp_test_dir.03M0vq/published_messages_from_multiple_topics_are_recorded.db3'.
[INFO] [rosbag2]: Waiting for messages...
Process 21800 stopped
* thread #2, stop reason = EXC_BAD_ACCESS (code=2, address=0x102a00000)
    frame #0: 0x0000000100f7cae5 libfastrtps.1.dylib`eprosima::fastrtps::ParticipantImpl::removePublisher(eprosima::fastrtps::Publisher*) [inlined] std::__1::pair<eprosima::fastrtps::Publisher*, eprosima::fastrtps::PublisherImpl*>::operator=(this=0x0000000102a00000, __p=0x0000000102a00010) at utility:509
   506 	        _NOEXCEPT_(is_nothrow_move_assignable<first_type>::value &&
   507 	                   is_nothrow_move_assignable<second_type>::value)
   508 	    {
-> 509 	        first = _VSTD::forward<first_type>(__p.first);
   510 	        second = _VSTD::forward<second_type>(__p.second);
   511 	        return *this;
   512 	    }
Target 0: (rosbag2_record_integration_test) stopped.
```

Karsten1987, 2018-09-28:
fixed in #32

closed 2018-09-28 by Karsten1987