ros2/rosbag2 #55: ros2 bag play - interaction

Karsten1987, 2018-10-29:
Like in ROS 1, there should be a chance to control the replaying of a bag file.

`spacebar` should toggle the start/pause replaying.
`s` should solely spin the bag once.

Generally, as for now, there is no status feedback given when replaying a bag file. As a first step, there should be a message printed, whether the bag file is currently publishing data or not.

weihunko, 2020-09-22:
Hi @Karsten1987 , while troubleshooting some perception issues with the recorded bag file, we would hope to have the functionalities to go back and forward a fixed amount of time, maybe 100ms or 1 second. 

For forwarding one step, we can use the `spin_once` function, but is it possible to go back one step or a fix amount of time?

Karsten1987, 2020-09-22:
I don't think I would temper with the `spin_once` functionality for going back and forth, more like with the "current" time step. 
See https://github.com/ros2/rosbag2/blob/56e440caca4bed3ea4cdde795a24a1bc60cdc355/rosbag2_transport/src/rosbag2_transport/player.cpp#L160

I imagine the overall behavior being the same, whether you'd like to step forward or backward. You'd simply add a `delta` to the current time and let the bag file play as usual.
Regardless of how you'd realize this though, there's the [issue of time (simulated) time](https://github.com/ros2/rosbag2/issues/99), which comes into play here as well, I assume.

weihunko, 2020-09-23:
@Karsten1987 From 
https://github.com/ros2/rosbag2/blob/56e440caca4bed3ea4cdde795a24a1bc60cdc355/rosbag2_transport/src/rosbag2_transport/player.cpp#L160

 It looks like `start_time_` is only used to control the publishing rate? Does it also used to determine which messages to publish according to their timestamps?

Kaju-Bubanja, 2020-12-02:
What's the status of this issue? Pausing rosbags was working fine in ros, does this feature not exists in ros2?

emersonknapp, 2020-12-07:
Not yet - it's still an open feature request

MichaelOrlov, 2021-03-12:
@emersonknapp @Karsten1987 I've implemented this new API aka `play`, `pause` and `play next in pause` in Apex.AI. \
We want to upstream our changes.  \
Could you please assign this issue to me?

emersonknapp, 2021-03-12:
Two slightly different approaches to this feature are being introduced by 
https://github.com/ros2/rosbag2/pull/675 - design for sim-time handling allows for pause/resume time
https://github.com/ros2/rosbag2/pull/666 - starting and stopping playback (but not time) by lifecycle node

We'd be happy to consider your approach as well, if you open a PR for it. Note that we prefer to review small incremental changes, as they are easier to review and validate.

MichaelOrlov, 2021-03-14:
Hi @emersonknapp Thank you for pointing out about other approaches. \
I am in the process of upstreaming my version to the branch to be able to create pull request. It turns out that our fork of the rosbag2 slightly different and it's not straight forward to cherry pick commits, I have to address changes manually and some time re-implement them to accommodate difference in base.  

As regards to the complexity and size of the delta. This feature looks simple, like add new API with a few methods, but it turns out this is not trivial in implementation. If want to replay deterministically in time domain with pause/resume and possibility to change speed and play next in pause. \
It's kind of challenging to implement and handling corner cases especially keep desired time interval between  messages after resuming with changed speed after played a few messages with play-next API. Also it was a quite challenging to implement it efficiently without locking mutexes in runtime and without affecting performance and avoiding race conditions and undefined behaviors. Also one important moment I was keeping in mind when implementing this feature is that errors in time delay (due to the some overhead) shouldn't  accumulate in sequence of playing messages and should be local for one inter-message interval. 

My approach probably more straight forward and simple in design rather than a couple previous ones but still a little bit complicated in understanding in some cases.

MichaelOrlov, 2021-03-16:
Hi @emersonknapp @Karsten1987 I've created [Pull request](https://github.com/ros2/rosbag2/pull/680). Could you please kindly review it?

closed 2021-09-27 by emersonknapp