ros2/rosbag2 #87: Errors when building rosbag2 storage plugins against released binaries

ruffsl, 2019-02-06:
I've been trying to build a rosbag2 storage plugin that is currently working with building against rosbag2 from source and have packaged separately from `rosbag2_storage_default_plugins`. However, when I try and build the same plugin package against the released binaries for crystal, I get some odd errors with colcon looking in the local workspace's `install` directory for `package.sh` files for packages that are already installed via the system. See the error except and reference dockerfile and plugin package below:

```
. /opt/ros/$ROS_DISTRO/setup.sh &&     apt-get update &&     rosdep install -q -y       --from-paths         src/rosbag2/rosbag2_storage_checkpoint_plugin       --ignore-src     && rm -rf /var/lib/apt/lists/*
...
executing command [apt-get install -y -qq ros-crystal-rosbag2-storage]
executing command [apt-get install -y -qq ros-crystal-rosbag2-storage-default-plugins]
executing command [apt-get install -y -qq ros-crystal-sqlite3-vendor]
executing command [apt-get install -y -qq ros-crystal-rosbag2-test-common]
#All required rosdeps installed successfully
Removing intermediate container a414f16f4c7b
 ---> 32bdfaafbdb0
Step 8/10 : ARG CMAKE_BUILD_TYPE=Release
 ---> Running in 3d7bcf664fc6
Removing intermediate container 3d7bcf664fc6
 ---> 110a131cfa19
Step 9/10 : RUN . /opt/ros/$ROS_DISTRO/setup.sh &&     colcon build       --symlink-install       --packages-select         rosbag2_storage_checkpoint_plugin       --cmake-args         -DCMAKE_BUILD_TYPE=$CMAKE_BUILD_TYPE
 ---> Running in 56a633e756e5
Starting >>> rosbag2_storage_checkpoint_plugin
[0.299s] ERROR:colcon.colcon_cmake.task.cmake.build:Failed to find the following files:
- /opt/ros_ws/install/rosbag2_test_common/share/rosbag2_test_common/package.sh
- /opt/ros_ws/install/sqlite3_vendor/share/sqlite3_vendor/package.sh
- /opt/ros_ws/install/rosbag2_storage/share/rosbag2_storage/package.sh
- /opt/ros_ws/install/rosbag2_storage_default_plugins/share/rosbag2_storage_default_plugins/package.sh
Check that the following packages have been built:
- rosbag2_test_common
- sqlite3_vendor
- rosbag2_storage
- rosbag2_storage_default_plugins
Failed   <<< rosbag2_storage_checkpoint_plugin	[ Exited with code 1 ]

Summary: 0 packages finished [0.17s]
  1 package failed: rosbag2_storage_checkpoint_plugin
```

<details>

Example package and Dockerfile:

https://github.com/ruffsl/rosbag2/tree/8dfcc290bb306ec606dcf56c7e798c63b98776ff/rosbag2_storage_checkpoint_plugin

``` Dockerfile
FROM ros:crystal

# install ROS2 dependencies
RUN apt-get update && apt-get install -q -y \
      build-essential \
      cmake \
      git \
      python3-colcon-common-extensions \
      python3-vcstool \
      wget \
    && rm -rf /var/lib/apt/lists/*

# copy ros package repo
ENV ROS_WS /opt/ros_ws
RUN mkdir -p $ROS_WS/src
WORKDIR $ROS_WS
COPY ./ src/rosbag2/

# install dependency package dependencies
RUN . /opt/ros/$ROS_DISTRO/setup.sh && \
    apt-get update && \
    rosdep install -q -y \
      --from-paths \
        src/rosbag2/rosbag2_storage_checkpoint_plugin \
      --ignore-src \
    && rm -rf /var/lib/apt/lists/*

# build dependency package source
ARG CMAKE_BUILD_TYPE=Release
RUN . /opt/ros/$ROS_DISTRO/setup.sh && \
    colcon build \
      --symlink-install \
      --packages-select \
        rosbag2_storage_checkpoint_plugin \
      --cmake-args \
        -DCMAKE_BUILD_TYPE=$CMAKE_BUILD_TYPE

# source navigation2 workspace from entrypoint
RUN sed --in-place \
      's|^source .*|source "$ROS_WS/install/setup.bash"|' \
      /ros_entrypoint.sh
```

dirk-thomas, 2019-02-06:
You are trying to only select a single package for building from the workspace without having built its dependencies. Since these dependencies are present in the workspace `colcon` assumes that you intended to use them. Hence the error message:

```
Check that the following packages have been built:
- rosbag2_test_common
- sqlite3_vendor
- rosbag2_storage
- rosbag2_storage_default_plugins
```

If you want to use the installed dependencies and not want to consider the dependencies which exist in the workspace you have the following options:
* remove the undesired dependencies from the workspace
* `--packages-ignore(-regex)` the packages from the workspace which you don't want to be considered (note the semantic difference between ignoring and selecting packages)
* place `COLCON.IGNORE` files into the source directories of the packages which you don't want to be considered
* pass `--paths <path/to/rosbag2_storage_checkpoint_plugin>` to only consider the package in that directory (not being aware what else is under `src`)

ruffsl, 2019-02-06:
> these dependencies are present in the workspace `colcon` assumes that you intended to use them.

Ah, this was perhaps the aspect I was not aware of.

> If you want to use the installed dependencies and not want to consider the dependencies which exist in the workspace you have the following options:

* I'd like to avoid having to constantly modify/curate the workspace file system at build time, so that hampers the use of `COLCON_IGNORE` (with an underscore, no dot, correct?) or deleting directories.
* I had to use `--base-paths` instead of `--paths` (what's the difference? build --help prompt makes them sound similar in where to search for packages) to limit colcon's focus on the respective package directory. Having to enumerate package paths wrt pwd is a bit cumbersome to automate.

I tried using `--packages-select` in concert with `--packages-ignore-regex` to allow for enumeration of only the packages I'd like to build, while ignoring the rest of the packages in the workspace all together:

```
colcon build \
      --symlink-install \
      --packages-select \
        rosbag2_storage_checkpoint_plugin \
      --packages-ignore-regex ".*"
```

But it seems that `--packages-ignore-regex` filter takes precedence, and thus skips over even the selected packages. So I guess there's no simple argumentative way yet to do this with package names alone?

dirk-thomas, 2019-02-06:
> But it seems that `--packages-ignore-regex` filter takes precedence, and thus skips over even the selected packages.

The `--packages-ignore` arguments are part of the package discovery phase which happens first. The `--packages-select` arguments are part of the package selection phase which happens after. So yes, the first one will ignore all packages and leave nothing to be selected.

> So I guess there's no argumentative way yet to do this with package names alone?

In your case you could use a regex which matches all package names but the one you would like to build in `--packages-ignore-regex`?

Or if the case gets more complicated you might want to use a two-step process using `colcon list` to determine your set of package names:

`colcon build --packages-ignore $(colcon list --names-only --packages-skip rosbag2_storage_checkpoint_plugin)`

ruffsl, 2019-02-06:
Thanks @dirk-thomas , using `colcon list` this way is a fine solution!

I'd still like to get this question, but I could make a separate ticket on colcon's docs if you'd prefer:

> `--base-paths` instead of `--paths` (what's the difference? build --help prompt makes them sound similar in where to search for packages)

closed 2019-02-06 by ruffsl

dirk-thomas, 2019-02-06:
> what's the difference? build --help prompt makes them sound similar in where to search for packages)

From the usage:

>  --base-paths [PATH [PATH ...]]
> > The base paths to recursively crawl for packages (default: .)

This one searches recursively for packages and is the default option.

>  --paths [PATH [PATH ...]]
> > The paths to check for a package. Use shell wildcards (e.g. `src/*`) to select all direct subdirectories

This one only checks the specific paths passes if they contain a package (without doing any recursive search). If you pass `src/*` then you shell will expand the wildcard argument into multi specific paths.