ros2/rosbag2 #92: Failed to load entry point 'play': cannot import name 'NODE_NAME_PREFIX'

tzven0, 2019-02-28:
Hey,

i was finally able to successfully build rosbag2 on aarch64 with

_colcon build --cmake-args -DBUILD_TESTING:BOOL=OFF_

since LeakSanitizer is not available for arm64 Ubuntu 16.04 Xenial.

Afterwards i get the following error.

```bash
$ ros2 bag --help
Failed to load entry point 'play': cannot import name 'NODE_NAME_PREFIX'
Failed to load entry point 'record': cannot import name 'NODE_NAME_PREFIX'
usage: ros2 bag [-h] Call `ros2 bag <command> -h` for more detailed usage. ...

Various rosbag related sub-commands

optional arguments:
  -h, --help            show this help message and exit

Commands:
  info  ros2 bag info

  Call `ros2 bag <command> -h` for more detailed usage.
```

Has anyone ever stumbled across this problem and knows how to solve it?

Thanks in advance!

Karsten1987, 2019-02-28:
Please make sure you have the latest version of `ros2cli` as well:
https://github.com/ros2/ros2cli/pull/158

tzven0, 2019-03-01:
Hello Karsten1987,

thanks for the reply. I tried the suggestion by cloning ros2cli and rebuilding.
Afterwards i get:

```
$ ros2 bag
Failed to load entry point 'play': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'record': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'start': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'status': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'stop': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'get': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'list': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'set': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'nodes': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'list': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'info': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'get': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'list': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'delete': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'set': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'generate_permissions': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'list': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'call': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'list': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'delay': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'hz': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'echo': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'pub': cannot import name 'HIDDEN_NODE_PREFIX'
Failed to load entry point 'info': cannot import name 'HIDDEN_NODE_PREFIX'
usage: ros2 bag [-h] Call `ros2 bag <command> -h` for more detailed usage. ...

Various rosbag related sub-commands

optional arguments:
  -h, --help            show this help message and exit

Commands:
  info  ros2 bag info

  Call `ros2 bag <command> -h` for more detailed usage.
```

Seems i am missing something substantial.

Thanks in advance!

Karsten1987, 2019-03-01:
did you rebuild your complete workspace?
It looks to me like you have a mismatch of package versions. I would recommend you check out the latest ros2 master repos and build it again.
https://github.com/ros2/ros2/blob/master/ros2.repos

tzven0, 2019-03-04:
Hey Karsten1987,

thanks for your suggestion. This actually solved the problem.

But, after installing from source, i found out i had to copy files within ros1_bridge installation folder.

ros1_storage_vendor.. failed since it missed the two .cmake files in install/ros1_bridge/share/ros1_bridge/cmake:

find_ros1_package.cmake
find_ros1_interface.cmake

I copied both files from src/ros2/ros1_bridge/cmake folder and it finally compiled.

Thanks again!

closed 2019-03-04 by tzven0

Karsten1987, 2019-03-04:
make sure you also have the latest `ros1bridge` repo. You might be missing the latest commit on this:
https://github.com/ros2/ros1_bridge/pull/164

tzven0, 2019-03-04:
I will also try that tomorrow and report :) Thanks again!