ros2/rosbag2 #94: ros2: Failed to load entry point - Segmentation fault

odel4y, 2019-03-21:
Hi,
I am running Ubuntu 18.04 with the latest Melodic and Crystal installed from the repositories. I am trying to use the rosbag_v2 plugin to read in a ROS1 rosbag. In a terminal where I sourced the Crystal installation the plugin is not found:
```
user@user-VirtualBox:~/host$ ros2 bag info -s rosbag_v2 amsterdamer09.bag
DISCLAIMER
ros2 bag is currently under development and not ready to use yet
[ERROR] [rosbag2_storage]: Could not load/open plugin with storage id 'rosbag_v2'.
[ERROR] [rosbag2_transport]: Could not read metadata for amsterdamer09.bag. Please specify the path to the folder containing an existing 'metadata.yaml' file or provide correct storage id if metadata file doesn't exist (see help).
```
If I source both Crystal and Melodic (in that order) I get a segmentation fault when using ros2cli:
```
user@user-VirtualBox:~/host$ source /opt/ros/melodic/setup.bash
ROS_DISTRO was set to 'crystal' before. Please make sure that the environment does not mix paths from different distributions.

user@user-VirtualBox:~/host$ ros2 bag info -s rosbag_v2 amsterdamer09.bag
Failed to load entry point 'info': /opt/ros/crystal/lib/librosbag2_storage.so: undefined symbol: _ZNK12class_loader11ClassLoader15isLibraryLoadedEv
The C extension '/opt/ros/crystal/lib/python3.6/site-packages/rosbag2_transport/_rosbag2_transport_py.cpython-36m-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Segmentation fault (core dumped)
```
I already followed the proposed link to the Wiki and reiterated the installation instructions for Linux, but that didn't help. I would be thankful for some advice.

Karsten1987, 2019-03-25:
I can reproduce this behavior. It looks like the latest debian packaging did not fully compile the `rosbag_v2_plugins` package correctly and thus the library for it can't be found.
I'll investigate this further. 

In the meantime, given that you play - what it looks like - your own ros1 bag file, we anyway recommend building the `ros1bridge` and this `rosbag_v2_plugins` package from source to make sure that all necessary message definitions are available during runtime.

AndreasAZiegler, 2019-04-06:
I'm facing the same issue.
@odel4y Did you make any progress.

Karsten1987, 2019-04-07:
These PRs (https://github.com/ros2-gbp/rosbag2-release/pull/3) should fix the problem when releasing the package the next time.

AndreasAZiegler, 2019-04-08:
@Karsten1987 Thanks for the information.

I removed ```ros-crystal-ros1-bridge``` and ```ros-crystal-ros1-rosbag-storage-vendor``` from my Ubuntu 18.04 system and built these two packages from source (current master branch). Unfortunately I still get  the same error

```
andreasziegler@AndreasZiegler:~/OpenSource/poly-exploration_ws$ ros2 bag info ~/ros/exploration_ws/experiment_1.bag 
Failed to load entry point 'info': /home/andreasziegler/OpenSource/poly-exploration_ws/install/rosbag2_storage/lib/librosbag2_storage.so: undefined symbol: _ZNK12class_loader11ClassLoader15isLibraryLoadedEv
The C extension '/home/andreasziegler/OpenSource/poly-exploration_ws/install/rosbag2_transport/lib/python3.6/site-packages/rosbag2_transport/_rosbag2_transport_py.cpython-36m-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Segmentation fault (core dumped)
```

Karsten1987, 2019-04-08:
I can reproduce this problem. I believe it's again a problem that the `ros1_bridge` pulls in the class loader implementation of ROS1 rather than ROS2. Can you try to see if `/home/andreasziegler/OpenSource/poly-exploration_ws/install/rosbag2_storage/lib/librosbag2_storage.so` is linked against class loader from melodic?
I thought we somehow workaround this behavior in here: https://github.com/ros2/ros1_bridge/commit/070600e86e67aa405d32435a4697db1983af30b1

AndreasAZiegler, 2019-04-08:
It is indeed the case. ldd gives me the following whether I source melodic first and crystal afterwards or the other way arround.

```
andreasziegler@AndreasZiegler:~/OpenSource/poly-exploration_ws/install/rosbag2_storage/lib$ ldd librosbag2_storage.so 
	linux-vdso.so.1 (0x00007ffe068de000)
	libament_index_cpp.so => /opt/ros/crystal/lib/libament_index_cpp.so (0x00007fb9d1806000)
	libconsole_bridge.so.0.4 => /opt/ros/crystal/lib/libconsole_bridge.so.0.4 (0x00007fb9d1601000)
	libclass_loader.so => /opt/ros/melodic/lib/libclass_loader.so (0x00007fb9d13da000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007fb9d11c6000)
	librcutils.so => /opt/ros/crystal/lib/librcutils.so (0x00007fb9d0fb5000)
	libyaml-cpp.so.0.6 => /opt/ros/crystal/opt/yaml_cpp_vendor/lib/libyaml-cpp.so.0.6 (0x00007fb9d0cbb000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fb9d0a9c000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fb9d0713000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fb9d04fb000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fb9d010a000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007fb9cff05000)
	libPocoFoundation.so.50 => /usr/lib/libPocoFoundation.so.50 (0x00007fb9cfb5c000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fb9d1d7a000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fb9cf7be000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fb9cf5ba000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fb9cf3b2000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fb9cf140000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fb9cef23000)
```

Karsten1987, 2019-04-08:
So I just dug around a little bit and found out that the order of sourcing is crucial here:

in a **fresh** terminal, the steps I did to make it work:

```
source /opt/ros/melodic/setup.bash
source /opt/ros/crystal/setup.bash
source /home/karsten/rosbag2_overlay_ws/install/local_setup.bash
```

If I now look at `ldd /home/karsten/rosbag2_overlay_ws/install/rosbag2_storage/lib/librosbag2_storage.so` I see that class loader is correctly found within the crystal installation.

With that, I am able to run the following command:
```
root@40eb766fe532:~/rosbag2_ws# ros2 bag info -s rosbag_v2 src/rosbag2/rosbag2_tests/resources/test_bag.bag
DISCLAIMER
ros2 bag is currently under development and not ready to use yet
[INFO] [rosbag2_bag_v2_plugins]: ROS 1 to ROS 2 type mapping is not available for topic 'rosout' which is of type 'rosgraph_msgs/Log'. Skipping messages of this topic when replaying.

Files:             test_bag.bag
Bag size:          8.3 KiB
Storage id:        rosbag_v2
Duration:          3.0s
Start:             Dec  5 2018 09:06:06.974 (1544000766.974)
End                Dec  5 2018 09:06:09.975 (1544000769.975)
Messages:          11
Topic information: Topic: rosout | Type: rosgraph_msgs/Log | Count: 5 | Serialization Format: rosbag_v2
                   Topic: string_topic | Type: std_msgs/String | Count: 3 | Serialization Format: rosbag_v2
                   Topic: int_topic | Type: std_msgs/Int32 | Count: 3 | Serialization Format: rosbag_v2
```

---

I ran into the same problem as you before just by not sourcing the workspaces correctly. 
Note here, that it's important to start from a fresh terminal. Let's say you sourced ros2 first, followed by ros1. If you now source ros2 again in the same terminal, your `LD_LIBRARY_PATH` will not be updated and ros2 won't be found first, because the path does already exist in the `LD_LIBRARY_PATH`. Does this make sense? Kudos to @dirk-thomas for helping out here!

AndreasAZiegler, 2019-04-09:
Interesting, I gave it another try and sourced in the order you recommenced but still end up with the class_loader library from ROS1 being used.

```
Starting docker container (fresh terminal)
andreasziegler@AndreasZiegler:~$ source /opt/ros/melodic/setup.bash
ROS_DISTRO was set to 'crystal' before. Please make sure that the environment does not mix paths from different distributions.
andreasziegler@AndreasZiegler:~$ source /opt/ros/crystal/setup.bash
ROS_DISTRO was set to 'melodic' before. Please make sure that the environment does not mix paths from different distributions.
andreasziegler@AndreasZiegler:~$ source OpenSource/poly-exploration_ws/install/local_setup.bash 
not found: "/home/andreasziegler/OpenSource/poly-exploration_ws/install/rosbag2_tests/share/rosbag2_tests/local_setup.sh"
not found: "/home/andreasziegler/OpenSource/poly-exploration_ws/install/rosbag2_tests/share/rosbag2_tests/local_setup.bash"
andreasziegler@AndreasZiegler:~$ ldd OpenSource/poly-exploration_ws/install/rosbag2_storage/lib/librosbag2_storage.so 
	linux-vdso.so.1 (0x00007ffd753e7000)
	libament_index_cpp.so => /opt/ros/crystal/lib/libament_index_cpp.so (0x00007f12e20bc000)
	libconsole_bridge.so.0.4 => /opt/ros/crystal/lib/libconsole_bridge.so.0.4 (0x00007f12e1eb7000)
	libclass_loader.so => /opt/ros/melodic/lib/libclass_loader.so (0x00007f12e1c90000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007f12e1a7c000)
	librcutils.so => /opt/ros/crystal/lib/librcutils.so (0x00007f12e186b000)
	libyaml-cpp.so.0.6 => /opt/ros/crystal/opt/yaml_cpp_vendor/lib/libyaml-cpp.so.0.6 (0x00007f12e1571000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f12e1352000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f12e0fc9000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f12e0db1000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f12e09c0000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f12e07bb000)
	libPocoFoundation.so.50 => /usr/lib/libPocoFoundation.so.50 (0x00007f12e0412000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f12e2630000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f12e0074000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f12dfe70000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f12dfc68000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f12df9f6000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f12df7d9000)
```

dirk-thomas, 2019-04-09:
You might want to investigate your environment (`env | grep LD_LIB`, `end | grep PREFIX_PATH`) before / after each command to see what is wrong with your environment.

AndreasAZiegler, 2019-04-09:
@dirk-thomas Thanks a lot for your help. It was indeed a problem with ```LD_LIB```. Somehow my docker container had already defined parts of the crystal environment.

Karsten1987, 2019-04-09:
Great you could figure it out. Related to the documentation: you can find it on the package related read me file: https://github.com/ros2/rosbag2/blob/master/rosbag2_bag_v2_plugins/README.md

closed 2019-04-09 by Karsten1987

rsasaki0109, 2019-07-24:
@Karsten1987
Following your advice and the issue #94, 
I succeeded to run rosbag2.
Thank you!

llu0120, 2020-08-13:
Hi @Karsten1987, I'm facing the same problem and look through the discussion above. May I know what is the `rosbag2_overlay_ws` for? I'm trying to play the rosbag in ROS2, please help thanks!

Karsten1987, 2020-08-17:
the ros1_bridge which you can install via debians only comes with a default set of supported message types. If you have a ROS1 bagfile which has non-standard (custom) messages types, you'd have to compile the bridge and the rosbag2_v2 package from source. To this, we usually refer to as an overlay workspace.

hunkyu, 2020-10-09:
I recompiled the ros2 dashing source code on ubuntu 18.04 and it also appeared 

Failed to load entry point 'info': /home/gzxsmapper/ros2_dashing/install/rosbag2_storage/lib/librosbag2_storage.so: undefined symbol: _ZNK12class_loader11ClassLoader15isLibraryLoadedEv
The C extension '/home/gzxsmapper/ros2_dashing/install/rosbag2_transport/lib/python3.6/site-packages/rosbag2_transport/_rosbag2_transport_py.cpython-36m-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-without-library-present-on-the-system' for possible solutions

shlokgoel, 2021-01-07:
Did you find a solution for your above problem? @hunkyu