ros2/rosbag2 #95: ROS2 bag record issue

Karsten1987, 2019-03-25:
connected to ros2/rclcpp#663

original issue post:

I have a custom ROS2 message, which I built and sourced correctly and it successfully publishes and subscribes on the node I create. However, when I try to record the bag files using,

`ros2 bag record -topicname`

it throws this error,

`Failed to record: Something went wrong loading the typesupport library for message type package_name/msgname. Library could not be found.`

I don't know why this error, is it because it is a custom message? Or is it because I need to link the custom messages somewhere?

Karsten1987, 2019-03-25:
@JoltonDsouza I just tried that locally on my machine and it seems to be working just fine. 

Please make sure that you `source`d all terminals correctly. Make sure that commands such as `ros2 topic echo /topicname` work correctly before running the `ros2 bag` command. The terminal in which you run the `record` has to be aware of all message packages in the system.

JoltonDsouza, 2019-03-25:
@Karsten1987 Thank you for the insight. I source every terminal each time I use a new one. The command,
`ros2 topic echo /topicname` shows the topic name on the terminal too, still it throws this error. One thing I forgot to mention was I am using Python and not Cpp for the messages and Publisher code. What are you using?

Karsten1987, 2019-03-26:
I've been trying out your example and it works fine for me (I had to fix a few things in your script though). 
Again, can you verify that when running your example, tools like `ros2 topic echo /test1` work correctly and in the same terminal your `ros2 bag record` fails with the typesupport error you reported?

JoltonDsouza, 2019-03-26:
@Karsten1987 I can confirm that the echo works just fine in the same terminal, it prints the message with the values I provide it with. Could you provide me the steps from the start how you build everything for the custom message?

Karsten1987, 2019-03-26:
a custom message is no different from any other ros2 message package. So you can basically take every existing message package and apply the same logic for your own message definitions.

Once you compiled your workspace with your custom message, you can just source this workspace and your message is available.

Then in terminal1:
```
<run your example script>
```
In terminal2 you can confirm that:
```
ros2 topic echo /test1
```
if the data gets correctly published, you can record them:
```
ros2 bag record /test1
```

That's all the steps I am doing.

JoltonDsouza, 2019-03-26:
@Karsten1987 Can you provide the steps of how you compile your custom messages with your workspace, i.e what steps you take to do it from scratch or if you have a reference? Also you mentioned you made some changes in the code, could you tell me what precisely? Thanks.

Karsten1987, 2019-03-26:
for a reference implementation of messages, just have a look at this package here: https://github.com/ros2/common_interfaces/tree/master/std_msgs

for your script:
```
from overlay_msgs.msg import SampleMsg

import rclpy
import time

class ROSPublisher():

    def __init__(self):
        rclpy.init(args=None)
        self.node = rclpy.create_node('publisher')
        print('creating node')
        self.publisher = self.node.create_publisher(SampleMsg, 'test1')

    def timer_callback(self):
        ros_msg = SampleMsg()
        ros_msg.first_name = 'David'
        ros_msg.last_name = 'White'
        ros_msg.address = 'California'
        ros_msg.age = 60
        ros_msg.salary = 150
        self.node.get_logger().info('Publishing first name as: "%s"' % ros_msg.first_name)
        self.publisher.publish(ros_msg)

    def loopbak(self):
        print('entering loop')
        while True:
            self.timer_callback()
            time.sleep(1)

if __name__ == "__main__":
    test = ROSPublisher()
    test.loopbak()
```

JoltonDsouza, 2019-03-26:
@Karsten1987 thank you, I'm trying to figure out the problem from scratch. Could you provide me the CMakeLists.txt and package.xml that you have edited for this particular message?

Karsten1987, 2019-03-26:
the  `CMakeLists.txt` and `package.xml` files are really the same as in the `std_msgs` package referenced in my previous comment. I only changed the file name.

JoltonDsouza, 2019-03-27:
@Karsten1987 I finally solved the problem, my `CMakeLists.txt` file looked different from the `std_msgs` package you referenced, I made the changes and made it look like the `CMakeLists.txt` in the `std_msgs` and built everything again. Works like a charm.
Thanks for your invaluable input 👍

Karsten1987, 2019-03-27:
great to hear you could figure it out!

closed 2019-03-27 by Karsten1987