ros2/rosbag2 #97: Build failing on test_msgs

capaulson, 2019-04-02:
The result is the same on both the Master and Crystal branch: 

```
CMake Error at CMakeLists.txt:81 (find_package):
  By not providing "Findtest_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "test_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "test_msgs" with
  any of the following names:

    test_msgsConfig.cmake
    test_msgs-config.cmake

  Add the installation prefix of "test_msgs" to CMAKE_PREFIX_PATH or set
  "test_msgs_DIR" to a directory containing one of the above files.  If
  "test_msgs" provides a separate development package or SDK, be sure it has
  been installed.

```

Any hints on test_msgs?

dirk-thomas, 2019-04-02:
You need to either install the Debian package (`ros-crystaltest-msgs`) or have the [package](https://github.com/ros2/rcl_interfaces/tree/master/test_msgs) in your workspace.

closed 2019-04-02 by dirk-thomas

JensLepers, 2019-11-08:
I have a similar issue:

https://answers.ros.org/question/337160/rosbag2-test_msgs-error/