ros2/rosbag2 #99: Simulated time (--clock) in rosbag2

AndreasAZiegler, 2019-04-11:
From ros 1 I'm used to the ```--clock``` option in order that the topics have time stamps as If the system would run right now. I couldn't find this option in the documentation of rosbag2. How can I activate simulated time with rosbag2?

Karsten1987, 2019-08-05:
Sorry for being so late on replying here. The concept of simulating time for rosbag2 is not yet realized. Therefore, there is currently no `--clock` option. 

I am going ahead and close this issue. If there is a follow up to this topic, please feel free to re-open it and continue the conversation.

IanTheEngineer, 2020-01-10:
Follow-on - @Karsten1987 & @jacobperron -  playing back a `/clock` from a bagfile is very much a core feature in the ROS1 `rosbag`. Otherwise, no node requiring time can utilize the collected bagfile data during playback. These nodes will use the system clock instead, which will conflict with the recorded data's header timestamps. Specifially, `tf` and nodes that subscribe to it are useless without `/clock`.

From a practical standpoint, I am running a high frequency system-clock-publishing-node publishing `/clock` onboard our robots simply to be recorded for bagfile purposes. This is pretty much the only workaround to debug data collected amcl, cartographer, perception nodes, etc.

scoopySHI, 2020-02-06:
Can't agree more @IanTheEngineer. With --clock publishing msg in current time should be a core feature in ros2bag since the nodes are using current system time. For instance, the robot_localization pkg cannot output correct prediction with "ros2 bag play" bagfile data. 
I have created a own [branch time_fix](https://github.com/scoopySHI/rosbag2.git) to solve the problem temporarily by deserialize the msg and replace the time stamper manually for topic name /imu/data. But this is definitely not the generic solution. It is not clear for me how to write the generic de_serialize process for all msg types. Hopefully it can be solved in Foxy.
Or maybe you can give us some hint how to solve it efficiently :) @dirk-thomas @ivanpauno @Karsten1987

kikass13, 2020-02-07:
@scoopySHI  +1

kikass13, 2020-02-13:
there is a solution from @scoopySHI which uses the ros2 message introspection to memcopy the "proper" ros time (message time propagated into current time) into the serialized message buffer.
That way we emulate a simulated time with the rosbag2 player. We do not know if this is the right way to do this, but it should work in principal and only requires introspection lookups for all messages before playing to find out if a header field exists.

This "feature" is obviously guarded by a --clock flag in the ros2bag python interface

Fork + branch:
https://github.com/scoopySHI/rosbag2/tree/time_fix

Relevant Section in source file:
https://github.com/scoopySHI/rosbag2/blob/time_fix/rosbag2_transport/src/rosbag2_transport/player.cpp#L191

This is a little buggy atm because we do not understand the message member offsets involved ... these structs define their offsets in a way which is not really consistent at first glance (because we do not properly understand the underlying dds implementation?). We will test it further tomorrow, but a little hint could help :)

kikass13, 2020-02-18:
see #299 for further details

kikass13, 2020-02-27:
our proposed solution did not work out in a way that was feasible.
**so this is still a problem.** 

There is now a drive to implement the `/clock` topic for simulating time like ros1 did. Apparently the implementation works on the node side.
 * how to do that properly is another issue and needs discussion 
    *  we have 3 separate solutions internally right now without even thinking too hard.

@scoopySHI and I will probably come back to this one at some point.

emersonknapp, 2020-03-27:
note: I rewrote the issue description at the top to add some details about how this should be implemented

vmanchal1, 2020-04-30:
Any chances this can make to the Foxy release?

emersonknapp, 2020-04-30:
PRs are welcome - I don't know of anybody that has time to work on it right now. I believe we're cutting the beta version of the package today for testing - @Karsten1987 does the beta count as a feature freeze or should we only freeze features at the release candidate phase?

Karsten1987, 2020-04-30:
I does not count for a feature freeze. I a feature like this one can be implemented in a somewhat straightforward way and we have still enough time to test that I am all up for it. I would just hold back from any major disruptive change across the code base at this point.

KenYN, 2020-08-27:
Just to add my tuppence-worth, with code like:

    timer_c_ = rclcpp::create_timer(
      this,
      this->get_clock(),
      rclcpp::Duration(std::chrono::milliseconds(timer_ms)),
	  std::bind(&MyClass::timerCallback_c, this));

This does not tick at all! I playback my v1 bagfile with:

    Ros1BagPlayNode = launch.actions.ExecuteProcess(
        cmd=['ros2', 'bag', 'play', '-s', 'rosbag_v2', LaunchConfiguration('bag_full_path')],
        name='rosbag_play',
        output='log',
        condition=IfCondition(PythonExpression(["'", LaunchConfiguration('bag_version'), "' == 'v1'"]))
    )

However, if I use the `ros1_bridge` and:

    rosbag play -r 0.5 --clock bag_full_path.bag

It works! This is a showstopper for us. At the simplistic level, if the `ros1_bridge` can do it, why can't ROS2 do it?

emersonknapp, 2020-08-31:
> At the simplistic level, if the ros1_bridge can do it, why can't ROS2 do it?

It's a functionality that exists in `rosbag`, to publish on the `/clock` topic on playback in sync with the messages being republished. This just hasn't been built in `rosbag2` yet - there is no fundamental blocker. PRs always welcome, of course :)

KenYN, 2020-09-01:
> This just hasn't been built in rosbag2 yet - there is no fundamental blocker.

I've looked at the code, and yes, I can see one way to do it, and perhaps the Play/Pause functionality should be added too. I'm not sure about a PR, though, as I'd have to get work's legal etc involved...
What happened about https://github.com/ros2/rosbag2/pull/300#issuecomment-589670688 ?

emersonknapp, 2020-09-04:
> What happened about [#300 (comment)](https://github.com/ros2/rosbag2/pull/300#issuecomment-589670688) ?

That PR unfortunately did not implement the desired feature. It was an attempt to re-stamp messages to current system time before publishing, rather than publishing to `/clock` to establish sim time. Additionally, in order to introspect and rewrite messages, it directly used an understanding of Fast-RTPS implementation, which made it nonportable to other RMW implementations.

dejanpan, 2020-12-17:
I just wanted to post here a workaround for this issue that we are using in one large [AD project](https://github.com/tier4/AutowareArchitectureProposal/tree/ros2): https://github.com/mitsudome-r/clock_publisher. 

This is JUST a workaround, the proper solution should be implemented as part of the ROS 2 G roadmap: https://index.ros.org/doc/ros2/Roadmap/#id2.

closed 2021-04-08 by emersonknapp

jediofgever, 2021-08-24:
sooo did this ever made to Foxy ?

emersonknapp, 2021-08-24:
No - there is currently no plan to backport this to Foxy due to the difficulty of maintaining API/ABI stability while doing so. It is available in Galactic and Rolling distributions. To use it in Foxy you might have some luck patching the https://github.com/ros2/rosbag2/tree/foxy-future branch, but I'm not sure

cristidotlumen, 2022-09-20:
For those who are still looking at this: the `--clock` option is not available in the foxy-future branch, but I got it working by starting from foxy-future, then use git cherry-pick with these commits (and fixing the conflicts): [PR for #689](https://github.com/ros2/rosbag2/commit/9ed91023f099436201cc9e4697291900063a6d63) and [PR for #695](https://github.com/ros2/rosbag2/commit/07c05da93761290a2b9e5a03ab733e094b63ebdd).

commit 009c60d5 (2021-03-31): Clock publisher in Player

Fixes #99
Design: #675
Depends on #689
Depends on #693 to expose to the CLI

Add a `rosgraph_msgs/Clock` publisher to the `Player` - that uses `PlayerClock::now()` to publish current time.
PR #695 (2021-04-08): /clock publisher in Player
Closes #99
Design: #675

Add a `rosgraph_msgs/Clock` publisher to the `Player` - that uses `PlayerClock::now()` to publish current time.

Expose this publish frequency to the CLI