ros2/rosbag2 #106: Build Fails on ROS2 Master

borgmanJeremy, 2019-05-02:
Hello,

I have my CI pipeline set up to build rosbag2 on the master release of ROS2. On 4/29 my build failed with the errors below. I started  digging into it and it think it might be related to this PR in rcl: https://github.com/ros2/rcl/pull/421

I started digging through it but its many layers of macro's so I am having a hard time pin pointing the exact problem. 

inished <<< sqlite3_vendor [2.43s]
Finished <<< rosbag2_test_common [3.64s]
Starting >>> rosbag2_storage
Finished <<< shared_queues_vendor [4.81s]
--- stderr: rosbag2_storage‌
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/ros_helper.cpp:54:11: error: format string is not a string literal (potentially insecure) [-Werror,-Wformat-security]
          ROSBAG2_STORAGE_LOG_ERROR_STREAM(
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/include/rosbag2_storage/logging.hpp:40:59: note: expanded from macro 'ROSBAG2_STORAGE_LOG_ERROR_STREAM'
    RCUTILS_LOG_ERROR_NAMED(ROSBAG2_STORAGE_PACKAGE_NAME, __ss.str().c_str()); \
                                                          ^~~~~~~~~~~~~~~~~~
/root/ros2_ws/ros2_base/install/include/rcutils/logging_macros.h:997:5: note: expanded from macro 'RCUTILS_LOG_ERROR_NAMED'
    __VA_ARGS__)
    ^~~~~~~~~~~
/root/ros2_ws/ros2_base/install/include/rcutils/logging_macros.h:72:64: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
      rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                               ^~~~~~~~~~~
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/ros_helper.cpp:54:11: note: treat the string as an argument to avoid this
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/include/rosbag2_storage/logging.hpp:40:59: note: expanded from macro 'ROSBAG2_STORAGE_LOG_ERROR_STREAM'
    RCUTILS_LOG_ERROR_NAMED(ROSBAG2_STORAGE_PACKAGE_NAME, __ss.str().c_str()); \
                                                          ^
1 error generated.
make[2]: *** [CMakeFiles/rosbag2_storage.dir/src/rosbag2_storage/ros_helper.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/storage_factory.cpp:24:
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/./impl/storage_factory_impl.hpp:92:7: error: format string is not a string literal (potentially insecure) [-Werror,-Wformat-security]
      ROSBAG2_STORAGE_LOG_ERROR_STREAM("Unable to create class load instance: " << e.what());
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/include/rosbag2_storage/logging.hpp:40:59: note: expanded from macro 'ROSBAG2_STORAGE_LOG_ERROR_STREAM'
    RCUTILS_LOG_ERROR_NAMED(ROSBAG2_STORAGE_PACKAGE_NAME, __ss.str().c_str()); \
                                                          ^~~~~~~~~~~~~~~~~~
/root/ros2_ws/ros2_base/install/include/rcutils/logging_macros.h:997:5: note: expanded from macro 'RCUTILS_LOG_ERROR_NAMED'
    __VA_ARGS__)
    ^~~~~~~~~~~
/root/ros2_ws/ros2_base/install/include/rcutils/logging_macros.h:72:64: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
      rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                               ^~~~~~~~~~~
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/./impl/storage_factory_impl.hpp:92:7: note: treat the string as an argument to avoid this
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/include/rosbag2_storage/logging.hpp:40:59: note: expanded from macro 'ROSBAG2_STORAGE_LOG_ERROR_STREAM'
    RCUTILS_LOG_ERROR_NAMED(ROSBAG2_STORAGE_PACKAGE_NAME, __ss.str().c_str()); \
                                                          ^
In file included from /root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/storage_factory.cpp:24:
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/./impl/storage_factory_impl.hpp:99:7: error: format string is not a string literal (potentially insecure) [-Werror,-Wformat-security]
      ROSBAG2_STORAGE_LOG_ERROR_STREAM("Unable to create class load instance: " << e.what());
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/include/rosbag2_storage/logging.hpp:40:59: note: expanded from macro 'ROSBAG2_STORAGE_LOG_ERROR_STREAM'
    RCUTILS_LOG_ERROR_NAMED(ROSBAG2_STORAGE_PACKAGE_NAME, __ss.str().c_str()); \
                                                          ^~~~~~~~~~~~~~~~~~
/root/ros2_ws/ros2_base/install/include/rcutils/logging_macros.h:997:5: note: expanded from macro 'RCUTILS_LOG_ERROR_NAMED'
    __VA_ARGS__)
    ^~~~~~~~~~~
/root/ros2_ws/ros2_base/install/include/rcutils/logging_macros.h:72:64: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
      rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                               ^~~~~~~~~~~
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/./impl/storage_factory_impl.hpp:99:7: note: treat the string as an argument to avoid this
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/include/rosbag2_storage/logging.hpp:40:59: note: expanded from macro 'ROSBAG2_STORAGE_LOG_ERROR_STREAM'
    RCUTILS_LOG_ERROR_NAMED(ROSBAG2_STORAGE_PACKAGE_NAME, __ss.str().c_str()); \
                                                          ^
In file included from /root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/storage_factory.cpp:24:
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/./impl/storage_factory_impl.hpp:112:7: error: format string is not a string literal (potentially insecure) [-Werror,-Wformat-security]
      ROSBAG2_STORAGE_LOG_ERROR_STREAM(
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/include/rosbag2_storage/logging.hpp:40:59: note: expanded from macro 'ROSBAG2_STORAGE_LOG_ERROR_STREAM'
    RCUTILS_LOG_ERROR_NAMED(ROSBAG2_STORAGE_PACKAGE_NAME, __ss.str().c_str()); \
                                                          ^~~~~~~~~~~~~~~~~~
/root/ros2_ws/ros2_base/install/include/rcutils/logging_macros.h:997:5: note: expanded from macro 'RCUTILS_LOG_ERROR_NAMED'
    __VA_ARGS__)
    ^~~~~~~~~~~
/root/ros2_ws/ros2_base/install/include/rcutils/logging_macros.h:72:64: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
      rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                               ^~~~~~~~~~~
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/./impl/storage_factory_impl.hpp:112:7: note: treat the string as an argument to avoid this
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/include/rosbag2_storage/logging.hpp:40:59: note: expanded from macro 'ROSBAG2_STORAGE_LOG_ERROR_STREAM'
    RCUTILS_LOG_ERROR_NAMED(ROSBAG2_STORAGE_PACKAGE_NAME, __ss.str().c_str()); \
                                                          ^
In file included from /root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/storage_factory.cpp:24:
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/src/rosbag2_storage/./impl/storage_factory_impl.hpp:131:7: error: format string is not a string literal (potentially insecure) [-Werror,-Wformat-security]
      ROSBAG2_STORAGE_LOG_ERROR_STREAM(
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/ros2_ws/ros2_ov/rosbag2/rosbag2_storage/include/rosbag2_storage/logging.hpp:40:59: note: expanded from macro 'ROSBAG2_STORAGE_LOG_ERROR_STREAM'
    RCUTILS_LOG_ERROR_NAMED(ROSBAG2_STORAGE_PACKAGE_NAME, __ss.str().c_str()); \

dirk-thomas, 2019-05-02:
To fix the build with the currrent master the following needs to be fixed / merged:
* #104
* #107
* @jacobperron update the tests using `test_msgs`

thomas-moulard, 2019-05-02:
@dirk-thomas I'm seeing a few warnings in `pluginlib`, taking a look at it now.

thomas-moulard, 2019-05-02:
Nevermind, #107 is fixing what I saw. LGTM'ed it as this fixes the issue on my machine.

closed 2019-05-07 by Karsten1987

borgmanJeremy, 2019-05-07:
Thanks this fixed my issue.

PR #111 (2019-05-07): fix compilation against master
Fixes #106.