ros2/rosbag2 #109: Enable play end-to-end tests when deterministic start is available

Karsten1987, 2019-05-03:
The current play end-to-end tests can be executed reliably as the publishers are sending their messages before any subscriber can be matched in a deterministic fashion. In order to push the repos further towards the ros2.repos integration, we disabled this test. See conversation here:
https://github.com/ros2/rosbag2/pull/104#issuecomment-487422345

This can be re-enabled when a more deterministic start of replaying a rosbag is available. One way to work towards this is to introduce a `--paused` option to start a rosbag, prepare all publishers but don't start publishing directly. This allows discovery to match with the respective subscribers to guarantee receiving all sent messages.

Karsten1987, 2020-04-02:
when re-enabling these tests, please add a "legacy" test for testing rosbags with versions <= 4.
Original comment here: https://github.com/ros2/rosbag2/pull/345#pullrequestreview-386853349

MichaelOrlov, 2025-05-07:
- Closing as an outdated, stale issue. We already solved a problem with determinism in the integration tests.

closed 2025-05-07 by MichaelOrlov