ros2/rosbag2 #119: test_record_all failing when run in parallel with other tests using colcon test

pbaughman, 2019-05-09:
When building ros2 from source, the rosbag2 test "test_record_all" apparently does not play nice with tests from other packages because it starts to pick up topics from other tests.

If you do 'colcon test' and other tests are allowed to run in parallel, the rosbag2_transport tests will fail

#### To reproduce
In a source workspace:
```
colcon build --packages-up-to rosbag2_transport
source install/setup.sh
while ./build/rosbag2_transport/test_record_all; do :; done
```

This will run the test in a loop, and it should pass every time.  The output from a good test looks something like this:
```
Running main() from gmock_main.cc
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from RecordIntegrationTestFixture
[ RUN      ] RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded
[INFO] [rosbag2_transport]: Listening for topics...
[INFO] [rosbag2_transport]: Subscribed to topic '/parameter_events'
[INFO] [rosbag2_transport]: Subscribed to topic '/chatter'
[INFO] [rosbag2_transport]: Subscribed to topic '/string_topic'
[INFO] [rosbag2_transport]: Subscribed to topic '/array_topic'
[       OK ] RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded (274 ms)
[----------] 1 test from RecordIntegrationTestFixture (274 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (274 ms total)
[  PASSED  ] 1 test.
```

Now, in another terminal, do
```
colcon test --packages-up-to rclcpp
```

You will probably see the rosbag2 transport tests fail with a message like:
```
Running main() from gmock_main.cc
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from RecordIntegrationTestFixture
[ RUN      ] RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded
[INFO] [rosbag2_transport]: Listening for topics...
[INFO] [rosbag2_transport]: Subscribed to topic '/parameter_events'
[INFO] [rosbag2_transport]: Subscribed to topic '/chatter'
[INFO] [rosbag2_transport]: Subscribed to topic '/string_topic'
[INFO] [rosbag2_transport]: Subscribed to topic '/array_topic'
[INFO] [rosbag2_transport]: Subscribed to topic '/rcl_test_subscription_nominal_string_chatter'
/workspace/src/rosbag2/rosbag2_transport/test/rosbag2_transport/test_record_all.cpp:44: Failure
Value of: recorded_messages
Expected: size is equal to 4
  Actual: { 16-byte object <50-A4 65-95 AD-7F 00-00 40-A4 65-95 AD-7F 00-00>, 16-byte object <E0-01 66-95 AD-7F 00-00 D0-01 66-95 AD-7F 00-00>,
 16-byte object <40-5A 32-95 AD-7F 00-00 30-5A 32-95 AD-7F 00-00>, 16-byte object <40-2F 45-95 AD-7F 00-00 30-2F 45-95 AD-7F 00-00>, 16-byte ob
ject <C0-1F 45-95 AD-7F 00-00 B0-1F 45-95 AD-7F 00-00> }, whose size 5 doesn't match
[  FAILED  ] RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded (281 ms)
[----------] 1 test from RecordIntegrationTestFixture (281 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (281 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded

 1 FAILED TEST
```

You can see from the fosbag2_transport INFO log message in the test, that it picked up an extra topic `/rcl_test_subscription_nominal_string_chatter`  This particular example came from [here](https://github.com/ros2/rcl/blob/2815b6876fb9dd528689ed9fa005c724f3aca782/rcl/test/rcl/test_subscription.cpp#L193) but in our job, I've also seen it conflict with other test topics
```
3: [INFO] [rosbag2_transport]: Listening for topics...
3: [INFO] [rosbag2_transport]: Subscribed to topic '/string_topic'
3: [INFO] [rosbag2_transport]: Subscribed to topic '/my_ns/parameter_events'
3: [INFO] [rosbag2_transport]: Subscribed to topic '/parameter_events'
3: [INFO] [rosbag2_transport]: Subscribed to topic '/my_ns/chatter'
3: [INFO] [rosbag2_transport]: Subscribed to topic '/chatter_composition'
3: [INFO] [rosbag2_transport]: Subscribed to topic '/array_topic'
3: [INFO] [rosbag2_transport]: Subscribed to topic '/logging_demo_count'
```

#### How to fix?
Some options:
  * Could we make the test assert that size of the recorded messages is <at least> 4 in case some other topics got sampled?
  * Is there a way to namespace the topics that the test is recording?
    * Related: can we tell rosbag2 to only record from /array_topic and /string_topic?
  * Could we set some DDS isolation for the test?

I can take a crack at this - but wanted to see if anybody had a preferred approach first.

pbaughman, 2019-05-09:
Something similar happens with test_record_all_no_discovery.cpp if I run the test in a loop in parallel with
```
colcon test --packages-select rcl
```
but that may be a different failure - I don't see messages about rosbag2 subscribing to extra topics

```
Running main() from gmock_main.cc
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from RecordIntegrationTestFixture
[ RUN      ] RecordIntegrationTestFixture.record_all_without_discovery_ignores_later_announced_topics
[INFO] [rosbag2_transport]: Listening for topics...
[INFO] [rosbag2_transport]: Subscribed to topic '/parameter_events'
/workspace/src/rosbag2/rosbag2_transport/test/rosbag2_transport/test_record_all_no_discovery.cpp:41: Failure
Value of: writer_->get_messages()
Expected: size is equal to 0
  Actual: { 16-byte object <10-E6 3E-E1 79-7F 00-00 00-E6 3E-E1 79-7F 00-00> }, whose size 1 doesn't match
[  FAILED  ] RecordIntegrationTestFixture.record_all_without_discovery_ignores_later_announced_topics (240 ms)
[----------] 1 test from RecordIntegrationTestFixture (240 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (240 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] RecordIntegrationTestFixture.record_all_without_discovery_ignores_later_announced_topics

 1 FAILED TEST
```

pbaughman, 2019-05-24:
I've made some progress fixing test_record_all by doing:
```
-  ASSERT_THAT(recorded_messages, SizeIs(4));                                       
+  ASSERT_GE(recorded_messages.size(), 4u);                                         
   auto string_messages = filter_messages<test_msgs::msg::Primitives>(
     recorded_messages, string_topic);
   auto array_messages = filter_messages<test_msgs::msg::StaticArrayPrimitives>(
     recorded_messages, array_topic);
-  ASSERT_THAT(string_messages, SizeIs(2));                                         
-  ASSERT_THAT(array_messages, SizeIs(2));                                          
-  EXPECT_THAT(string_messages[0]->string_value, Eq("Hello World"));                
-  EXPECT_THAT(array_messages[0]->bool_values, ElementsAre(true, false, true));     
-  EXPECT_THAT(array_messages[0]->string_values,                                    
-    ElementsAre("Complex Hello1", "Complex Hello2", "Complex Hello3"));            
+  ASSERT_GE(string_messages.size(), 2u);                                           
+  ASSERT_GE(array_messages.size(), 2u);                                            
+                                                                                   
+                                                                                   
+  EXPECT_THAT(                                                                     
+    string_messages,                                                               
+    UnorderedElementsAre(                                                          
+      Pointee(Field(&test_msgs::msg::Primitives::string_value, Eq("Hello World"))),
+      Pointee(Field(&test_msgs::msg::Primitives::string_value, Eq("Hello World"))) 
+    )                                                                              
+  );                                                                               
+                                                                                   
+  EXPECT_THAT(                                                                     
+    array_messages,                                                                
+    UnorderedElementsAre(                                                          
+      Pointee(                                                                     
+        Field(                                                                     
+          &test_msgs::msg::StaticArrayPrimitives::bool_values,                     
+          ElementsAre(true, false, true)                                           
+        )                                                                          
+      ),                                                                           
+      Pointee(                                                                     
+        Field(                                                                     
+          &test_msgs::msg::StaticArrayPrimitives::bool_values,                     
+          ElementsAre(true, false, true)                                           
+        )                                                                          
+      )                                                                            
+    )                                                                              
+  );                                                                               
+                                                                                   
+  EXPECT_THAT(                                                                     
+    array_messages,                                                                
+    UnorderedElementsAre(                                                          
+      Pointee(                                                                     
+        Field(                                                                     
+          &test_msgs::msg::StaticArrayPrimitives::string_values,                   
+          ElementsAre("Complex Hello1", "Complex Hello2", "Complex Hello3")        
+        )                                                                          
+      ),                                                                           
+      Pointee(                                                                     
+        Field(                                                                     
+          &test_msgs::msg::StaticArrayPrimitives::string_values,                   
+          ElementsAre("Complex Hello1", "Complex Hello2", "Complex Hello3")        
+        )                                                                          
+      )                                                                            
+    )                                                                              
+  );                                                                               
```

But the interference goes both ways.  The rcl test `test_get_node_names` for example does not like to run at the same time as the rosbag2 tests

Karsten1987, 2019-05-28:
I believe it makes sense to have the rosbag2 tests run more robustly, but tests in general will always be subject to parallel execution when testing discovery behavior. 
In this case I assume it's fine to prefix topics, or namespace them, so avoid duplication and assert correctly to see that we have the right topics present.

I am happy to review a PR for this.

pbaughman, 2019-05-28:
@Karsten1987 I got about halfway done with fixing this before I realized that the interference goes both ways.  Initially I (naively) thought that other tests were making rosbag2 tests fail, but that rosbag2 testss didn't make other tests fail.  This isn't assumption is wrong.

I'm considering a more general approach now where we can use ROS_DOMAIN_ID or something else to isolate parallel tests from one-another.  I'm going to close this issue for now because while I'm confident we can 'fix' the rosbag2 tests, it will be very time consuming and maintenance  heavy.

closed 2019-05-28 by pbaughman