ros2/rosbag2 #126: More intuitive rosbag write API

eric1221bday, 2019-06-05:
I've been looking into potentially using rosbag2 to convert something like the KITTI dataset into a rosbag2 format. There are two ways to go about it, one is to convert and publish the data then use ros2bag record to record it, the other is to simply do the conversion and write directly to the bag using the rosbag2 API.

I might be mistaken but it seems that currently rosbag2 does not expose an intuitive interface for easy writing of rosbags. Looking at the writer itself for the record functionality, the API directly writes the serialized message into the bag. There should be, similar to the rosbag1 API, a way to simply pass in the deserialized msg directly and have the API handle the serialization and the writing.

Karsten1987, 2019-08-05:
@eric1221bday I totally agree with you. The C++ API is still up for improvement and also not set in stone. So if you have concrete ideas for a better API, please feel free to voice them here on this ticket.

eric1221bday, 2019-09-22:
So first off, should this be a separate API completely? Or an extension of the existing SequentialReader API? 

If an extension that obviously is simpler to implement but OTOH that might bloat the SequentialReader class.

Another consideration is how generic this API should be. The new rosbag2 sqlite3 format brings with it a lot of advantages such as random access, should this new API take advantage of that fact? If so, then it would have to ditch support for the old rosbags.

cmpute, 2019-10-22:
Is there any plan to port ROS1 like API for rosbag reading and writing [as described in the docs](http://wiki.ros.org/rosbag/Code%20API)? I think the API for the old rosbag files are pretty intuitive and easy to use. Maybe APIs with similar functionalities can be really helpful with better support for random access!

Karsten1987, 2020-01-29:
> The new rosbag2 sqlite3 format brings with it a lot of advantages such as random access, should this new API take advantage of that fact? If so, then it would have to ditch support for the old rosbags.

We've been just talking offline about having the capabilities to filter out topics while reading. So I could see an enhancement to set a generic `rosbag2_cpp::FilterObject` or something to the reader instances which can handle the `has_next()` and `read_next()` accordingly. Such a filter can be either a list of topic names or a specific time interval etc.
I further see it in the responsibility of `rosbag2_cpp`'s API to deal with this and each individual plugin should only adhere to a standard set of interface functions. If a plugin, such as old rosbags can't provide that functionality, they should handle appropriately by returning a warning or a default option.

---

> Is there any plan to port ROS1 like API for rosbag reading and writing as described in the docs?

I don't think we can easily mimic the ROS1 API. The `rosbag2_cpp` API has currently no dependency on the message packages and can thus only store serialized data. Which is intended so far. 
The way we can use the current API for writing bag files is pretty straight forward though and is something along these lines:

```c++
std_msgs::msg::String msg;
msg.data = "Hello World";

auto serialized_msg = rmw_get_zero_initialized_serialized_message();
auto allocator = rcutils_get_default_allocator();
auto initial_capacity = 0u;
auto ret = rmw_serialized_message_init(
     &serialized_msg,
     initial_capacity,
     &allocator);

auto string_ts =
      rosidl_typesupport_cpp::get_message_type_support_handle<std_msgs::msg::String>();
// Given the correct typesupport, we can convert our ROS2 message into
// its binary representation (serialized_msg)
ret = rmw_serialize(msg.get(), string_ts, &serialized_msg);

rosbag2_storage::SerializedBagMessage bag_message;
bag_message.serialized_data = serialized_msg.buffer;
bag_message.time_point = // time
bag_message.topic = // topic

// set up bag 
...

// write bag
rosbag2_cpp::write(bag_message);
```

The reason why we can't easily do the ROS1 approach is because of the different message type supports needed for doing the serialization. We essentially need to link against each message package in order to enable an API such as:
```c++
template<class MsgT>
rosbag2_cpp::write(std::string topic, rclcpp::TImePoint, MsgT message);
```

forever3000, 2020-06-04:
> The reason why we can't easily do the ROS1 approach is because of the different message type supports needed for doing the serialization. We essentially need to link against each message package in [...]
Do you have any plan to release these kinds of Ros2 API?

berndpfrommer, 2021-04-14:
Would be great to have a documented C++ API or even a simple working example of how to write a rosbag in ROS2. Is there an ETA for that? Thanks!

Karsten1987, 2021-04-15:
We've addressed some of these things lately:
Have a look at this: https://github.com/ros2/rosbag2/blob/master/rosbag2_tests/test/rosbag2_tests/test_rosbag2_cpp_api.cpp#L49-L87

I think that looks already kind of like we've envisioned during this ticket and I wonder whether we could close this ticket.

berndpfrommer, 2021-04-15:
I tried lifting that example out but the compiler croaked on the line with rosbag2_cpp::Writer()
```
rcpputils::fs::remove_all(rosbag_directory);

  {
    rosbag2_cpp::Writer writer;
    writer.open(rosbag_directory.string());

```
The foxy headers declare the constructor like so:
```
class ROSBAG2_CPP_PUBLIC Writer final
{
public:
  explicit Writer(std::unique_ptr<rosbag2_cpp::writer_interfaces::BaseWriterInterface> writer_impl);

  ~Writer();
```

emersonknapp, 2021-04-15:
> The foxy headers declare the constructor like so:

Already-released distros have an intentionally frozen API - changes like this one will only be available in development or on new distro releases, such as Galactic coming out next month.

Karsten1987, 2021-05-11:
> I realize that the actual crucial piece that I mentioned, the ability to directly pass deserialized messages to the writer, is the part that's still missing.

Isn't that what's depicted here: https://github.com/ros2/rosbag2/blob/master/rosbag2_tests/test/rosbag2_tests/test_rosbag2_cpp_api.cpp#L78-L81

emersonknapp, 2021-05-11:
> Isn't that what's depicted here:

I think what they are looking for is actually line 84 https://github.com/ros2/rosbag2/blob/master/rosbag2_tests/test/rosbag2_tests/test_rosbag2_cpp_api.cpp#L84 - calling `write` with an arbitrary message type like `test_msgs::msg::BasicTypes`

Karsten1987, 2021-05-11:
yeah, you're right. I misread. Thanks for the clarification.

emersonknapp, 2021-06-15:
Based on above conversation I am closing this as completed. Please open a new issue for any new specific feature requests

closed 2021-06-15 by emersonknapp