ros2/rosbag2 #129: Support for latching

DensoADAS, 2019-06-11:
**Required Info:**
- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - Dashing
- DDS implementation:
  - Fast-RTPS

**Expected behavior**
I would be nice if messages published with the durability profile `transient_local` would be recorded with a subscriber that also uses the durability profile `transient_local`, so that messages published before the start of the recording get also recorded. This messages should then also be played back using the profile `transient_local` in order to keep up the latching behavior.

Karsten1987, 2019-06-17:
This is related to #125 in a way that there is currently no direct way to obtain the QoS parameters solely given a topic name. The subscriber has to know the correct QoS settings a priori. 

One workaround, same as for unreliable connections is to introduce keywords within the command line tool to manually specify which topics are reliable, unreliable or latched.
Something like:

```
ros2 bag record <reliable_topic1> <reliable_topic2> --unreliable <unreliable_topic1> <unreliable_topic2> --latched <latched_topic1> <latched_topic2>
```

Does this sound feasible to you?

kfabian, 2019-06-19:
Thanks
That is probably a good workaround, since there seems to be no other solution until [https://github.com/ros2/rmw/issues/151](url) is resolved. But it would still be important to have this feature in the end.

DensoADAS, 2019-06-19:
Thanks
That is probably a good workaround, since there seems to be no other solution until https://github.com/ros2/rmw/issues/151 is resolved. But it would still be important to have this feature in the end.

emersonknapp, 2020-01-29:
This functionality will be handled by #125, closing as duplicate

closed 2020-01-29 by emersonknapp