ros2/rosbag2 #134: compilation issue with rosbag

vijaypalvai, 2019-06-22:
While compiling the rosbag on crystal using the instruction on the homepage, i get the following error. trying this on docker builds with ubuntu 18. can any one suggest whats going wrong here. 

/types/introspection_message.cpp: In function âvoid rosbag2::cleanup_element(void*, const MessageMember&)â:
/root/ros2_ws/src/rosbag2/rosbag2/src/rosbag2/types/introspection_message.cpp:92:71: error: âROS_TYPE_WSTRINGâ is not a member of ârosidl_typesupport_introspection_cppâ
   } else if (member.type_id_ == rosidl_typesupport_introspection_cpp::ROS_TYPE_WSTRING) {

Karsten1987, 2019-08-05:
The support for wide strings (`ROS_TYPE_WSTRING`) was introduced in dashing:
https://index.ros.org/doc/ros2/Releases/Release-Dashing-Diademata/#new-features-in-this-ros-2-release

In order to compile it correctly for crystal, you may want to make sure that all your ros2 workspace is on crystal.

Karsten1987, 2019-08-21:
@vijaypalvai can this ticket be closed or do you have further questions on this?