ros2/rosbag2 #138: Observed rosbag2_transport test_record hanging

pbaughman, 2019-06-27:
I observed this failure in CI, and again on my machine.  I was able to get a stack trace from GDB by doing

```
terminal 1> colcon build --packages-up-to rosbag2_transport
terminal 1> source install/setup.bash
terminal 1> while build/rosbag2_transport/test_record; do :; done  # Loop until the test fails, or in this case hangs

terminal 2> sudo gdb -p 1885  # The PID of the running test. . 
(gdb) where
```

The stack trace is as follows:
```
#0  0x00007f9c6b11998d in pthread_join (threadid=140308771751680, thread_return=0x0) at pthread_join.c:90
#1  0x00007f9c6a0fcb97 in std::thread::join() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00000000004e7d23 in std::_Mem_fn_base<void (std::thread::*)(), true>::operator()<, void>(std::thread&) const (this=0x7ffeb637a6e8, __object=...) at /usr/include/c++/5/functional:583
#3  0x00000000004d4286 in std::_Mem_fn_base<void (std::thread::*)(), true>::operator()<std::thread, , void>(std::reference_wrapper<std::thread>) const (this=0x7ffeb637a6e8, __ref=...) at /usr/include/c++/5/functional:619
#4  0x00000000004bb211 in std::_Bind_simple<std::_Mem_fn<void (std::thread::*)()> (std::reference_wrapper<std::thread>)>::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=0x7ffeb637a6e0) at /usr/include/c++/5/functional:1531
#5  0x000000000049dc56 in std::_Bind_simple<std::_Mem_fn<void (std::thread::*)()> (std::reference_wrapper<std::thread>)>::operator()() (this=0x7ffeb637a6e0) at /usr/include/c++/5/functional:1520
#6  0x000000000047d5aa in std::__once_call_impl<std::_Bind_simple<std::_Mem_fn<void (std::thread::*)()> (std::reference_wrapper<std::thread>)> >() () at /usr/include/c++/5/mutex:706
#7  0x00007f9c6b11fa99 in __pthread_once_slow (once_control=0x41f8578, init_routine=0x44ed70 <__once_proxy@plt>) at pthread_once.c:116
#8  0x00000000004501d0 in __gthread_once (__once=0x41f8578, __func=0x44ed70 <__once_proxy@plt>) at /usr/include/x86_64-linux-gnu/c++/5/bits/gthr-default.h:699
#9  0x0000000000467da2 in std::call_once<void (std::thread::*)(), std::reference_wrapper<std::thread> >(std::once_flag&, void (std::thread::*&&)(), std::reference_wrapper<std::thread>&&) (__once=..., __f=<unknown type in /home/pete.baughman/ws/build/rosbag2_transport/test_record, CU 0x0, DIE 0xd4d17>)
    at /usr/include/c++/5/mutex:738
#10 0x0000000000460b90 in std::__future_base::_Async_state_commonV2::_M_join (this=0x41f8550) at /usr/include/c++/5/future:1638
#11 0x0000000000460af4 in std::__future_base::_Async_state_commonV2::_M_complete_async (this=0x41f8550) at /usr/include/c++/5/future:1636
#12 0x000000000045fd11 in std::__future_base::_State_baseV2::wait (this=0x41f8550) at /usr/include/c++/5/future:319
#13 0x0000000000466ecf in std::__basic_future<void>::_M_get_result (this=0x386f240) at /usr/include/c++/5/future:681
#14 0x000000000046096f in std::future<void>::get (this=0x386f240) at /usr/include/c++/5/future:846
#15 0x0000000000461158 in rosbag2_test_common::PublisherManager::run_publishers(std::function<unsigned long (std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)>) (this=0x2415a08, count_function=...)
    at /home/pete.baughman/ws/install/include/rosbag2_test_common/publisher_manager.hpp:77
#16 0x0000000000463531 in RecordIntegrationTestFixture::run_publishers (this=0x2415910) at path/rosbag2_transport/test/rosbag2_transport/record_integration_fixture.hpp:68
#17 0x0000000000459308 in RecordIntegrationTestFixture_published_messages_from_multiple_topics_are_recorded_Test::TestBody (this=0x2415910) at path/rosbag2_transport/test/rosbag2_transport/test_record.cpp:46
#18 0x00000000005b4e3a in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void> (object=0x2415910, method=&virtual testing::Test::TestBody(), location=0x600c0b "the test body") at /home/pete.baughman/ws/install/src/gtest_vendor/src/gtest.cc:2447
#19 0x00000000005a8a3d in testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void> (object=0x2415910, method=&virtual testing::Test::TestBody(), location=0x600c0b "the test body") at /home/pete.baughman/ws/install/src/gtest_vendor/src/gtest.cc:2483
#20 0x000000000055e6ba in testing::Test::Run (this=0x2415910) at /home/pete.baughman/ws/install/src/gtest_vendor/src/gtest.cc:2523
#21 0x000000000055fc4a in testing::TestInfo::Run (this=0x2414220) at /home/pete.baughman/ws/install/src/gtest_vendor/src/gtest.cc:2703
#22 0x0000000000560978 in testing::TestCase::Run (this=0x2415330) at /home/pete.baughman/ws/install/src/gtest_vendor/src/gtest.cc:2825
#23 0x0000000000579e88 in testing::internal::UnitTestImpl::RunAllTests (this=0x2414f70) at /home/pete.baughman/ws/install/src/gtest_vendor/src/gtest.cc:5226
#24 0x00000000005b6a8c in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (object=0x2414f70, method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x57995a <testing::internal::UnitTestImpl::RunAllTests()>, 
    location=0x601610 "auxiliary test code (environments or event listeners)") at /home/pete.baughman/ws/install/src/gtest_vendor/src/gtest.cc:2447
#25 0x00000000005aaa18 in testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (object=0x2414f70, method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x57995a <testing::internal::UnitTestImpl::RunAllTests()>, 
    location=0x601610 "auxiliary test code (environments or event listeners)") at /home/pete.baughman/ws/install/src/gtest_vendor/src/gtest.cc:2483
#26 0x00000000005770f1 in testing::UnitTest::Run (this=0x984680 <testing::UnitTest::GetInstance()::instance>) at /home/pete.baughman/ws/install/src/gtest_vendor/src/gtest.cc:4834
#27 0x000000000054cf86 in RUN_ALL_TESTS () at /home/pete.baughman/ws/install/src/gtest_vendor/include/gtest/gtest.h:2372
#28 0x000000000054ce55 in main (argc=1, argv=0x7ffeb637b2f8) at /home/pete.baughman/ws/install/src/gmock_vendor/src/gmock_main.cc:53
```

I've got the following threads:
```
(gdb) info threads                                                                                                                                         
  Id   Target Id         Frame                                                                                                                             
* 1    Thread 0x7f9c6c6caf40 (LWP 1885) "test_record" 0x00007f9c6b11998d in pthread_join (threadid=140308771751680, thread_return=0x0) at pthread_join.c:90
  2    Thread 0x7f9c64076700 (LWP 1886) "test_record" 0x00007f9c6b120827 in futex_abstimed_wait_cancelable (private=0, abstime=0x0, expected=0,            
    futex_word=0x7f9c6bf09780 <rclcpp::SignalHandler::signal_handler_sem_>) at ../sysdeps/unix/sysv/linux/futex-internal.h:205                             
  3    Thread 0x7f9c63875700 (LWP 1887) "test_record" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185       
  4    Thread 0x7f9c63074700 (LWP 1888) "test_record" 0x00007f9c69b6ba13 in epoll_wait () at ../sysdeps/unix/syscall-template.S:84                         
  5    Thread 0x7f9c62873700 (LWP 1889) "test_record" 0x00007f9c6b12194d in recvmsg () at ../sysdeps/unix/syscall-template.S:84                            
  6    Thread 0x7f9c62072700 (LWP 1890) "test_record" 0x00007f9c6b12194d in recvmsg () at ../sysdeps/unix/syscall-template.S:84                            
  7    Thread 0x7f9c61871700 (LWP 1891) "test_record" 0x00007f9c6b12194d in recvmsg () at ../sysdeps/unix/syscall-template.S:84                            
  8    Thread 0x7f9c61070700 (LWP 1892) "test_record" pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185       
  9    Thread 0x7f9c53fff700 (LWP 1893) "test_record" 0x00007f9c69b6ba13 in epoll_wait () at ../sysdeps/unix/syscall-template.S:84                         
  10   Thread 0x7f9c537fe700 (LWP 1894) "test_record" 0x00007f9c6b12194d in recvmsg () at ../sysdeps/unix/syscall-template.S:84                            
  11   Thread 0x7f9c52ffd700 (LWP 1895) "test_record" 0x00007f9c6b12194d in recvmsg () at ../sysdeps/unix/syscall-template.S:84                            
  12   Thread 0x7f9c527fc700 (LWP 1896) "test_record" 0x00007f9c6b12194d in recvmsg () at ../sysdeps/unix/syscall-template.S:84                            
[... cut ...]
  16   Thread 0x7f9c2f7fe700 (LWP 1901) "test_record" 0x00007f9c6b12194d in recvmsg () at ../sysdeps/unix/syscall-template.S:84                            
  17   Thread 0x7f9c2e7fc700 (LWP 1903) "test_record" 0x00007f9c6b121c1d in nanosleep () at ../sysdeps/unix/syscall-template.S:84  
```

I've got version d630f8e right now, because we're not all the way up to date with dashing.  I didn't see anything in the changelog or open issues that indicates this has been noticed before.  I'm going to try to reproduce this on the nightly docker image next.

pbaughman, 2019-06-28:
When using faa22cc I wasn't able to get it to hang, but I can get it to segfault and drop a core file.  The stack trace is pretty deep:

```
(gdb) where
#0  0x00007f8e9daccd09 in hash_map_find (hash_map=hash_map@entry=0x7f8e9e32f590 <__logger_map>, key=key@entry=0x7f8e975a3cc8, key_hash=key_hash@entry=0x7f8e975a3c88, map_index=map_index@entry=0x7f8e975a3c90,
    bucket_index=bucket_index@entry=0x7f8e975a3c98, entry=entry@entry=0x7f8e975a3ca0) at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rcutils/src/hash_map.c:347
#1  0x00007f8e9dacdc11 in rcutils_hash_map_key_exists (hash_map=hash_map@entry=0x7f8e9e32f590 <__logger_map>, key=key@entry=0x7f8e975a3cc8) at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rcutils/src/hash_map.c:460
#2  0x00007f8e9e1133a7 in rcl_logging_rosout_init_publisher_for_node (node=node@entry=0x7f8e80000f90) at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rcl/rcl/src/rcl/logging_rosout.c:156
#3  0x00007f8e9e1158c2 in rcl_node_init (node=node@entry=0x7f8e80000f90, name=0x7f8e975ac770 "_rosbag2", namespace_=<optimized out>, context=0x55df91682e80, options=options@entry=0x7f8e80000d70)
    at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rcl/rcl/src/rcl/node.c:351
#4  0x00007f8e9e3de5f2 in rclcpp::node_interfaces::NodeBase::NodeBase (this=0x7f8e80000e30, node_name="_rosbag2", namespace_="", context=..., rcl_node_options=..., use_intra_process_default=<optimized out>)
    at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rclcpp/rclcpp/src/rclcpp/node_interfaces/node_base.cpp:66
#5  0x00007f8e9e3d7f3e in rclcpp::Node::Node (this=0x7f8e80000b30, node_name="_rosbag2", namespace_="", options=...) at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rclcpp/rclcpp/src/rclcpp/node.cpp:116
#6  0x00007f8e9e3d9b21 in rclcpp::Node::Node (this=0x7f8e80000b30, node_name="_rosbag2", options=...) at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rclcpp/rclcpp/src/rclcpp/node.cpp:103
#7  0x00007f8e9e8a5c39 in rosbag2_transport::Rosbag2Node::Rosbag2Node (this=0x7f8e80000b30, node_name="_rosbag2")
    at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rosbag2/rosbag2_transport/src/rosbag2_transport/rosbag2_node.cpp:33
#8  0x00007f8e9e8aa1a7 in __gnu_cxx::new_allocator<rosbag2_transport::Rosbag2Node>::construct<rosbag2_transport::Rosbag2Node, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (
    this=<optimized out>, __p=<optimized out>) at /usr/include/c++/7/ext/new_allocator.h:136
#9  std::allocator_traits<std::allocator<rosbag2_transport::Rosbag2Node> >::construct<rosbag2_transport::Rosbag2Node, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (__a=...,
    __p=<optimized out>) at /usr/include/c++/7/bits/alloc_traits.h:475
#10 std::_Sp_counted_ptr_inplace<rosbag2_transport::Rosbag2Node, std::allocator<rosbag2_transport::Rosbag2Node>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (__a=..., this=0x7f8e80000b20) at /usr/include/c++/7/bits/shared_ptr_base.h:526
#11 std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<rosbag2_transport::Rosbag2Node, std::allocator<rosbag2_transport::Rosbag2Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >    (__a=..., this=<optimized out>) at /usr/include/c++/7/bits/shared_ptr_base.h:637
#12 std::__shared_ptr<rosbag2_transport::Rosbag2Node, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rosbag2_transport::Rosbag2Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (
    __a=..., __tag=..., this=<optimized out>) at /usr/include/c++/7/bits/shared_ptr_base.h:1295
#13 std::shared_ptr<rosbag2_transport::Rosbag2Node>::shared_ptr<std::allocator<rosbag2_transport::Rosbag2Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (__a=..., __tag=...,
    this=<optimized out>) at /usr/include/c++/7/bits/shared_ptr.h:344
#14 std::allocate_shared<rosbag2_transport::Rosbag2Node, std::allocator<rosbag2_transport::Rosbag2Node>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (__a=...)
    at /usr/include/c++/7/bits/shared_ptr.h:691
#15 std::make_shared<rosbag2_transport::Rosbag2Node, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > () at /usr/include/c++/7/bits/shared_ptr.h:707
#16 rosbag2_transport::Rosbag2Transport::setup_node (this=this@entry=0x7f8e975ad090, node_prefix="")
    at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rosbag2/rosbag2_transport/src/rosbag2_transport/rosbag2_transport.cpp:82
#17 0x00007f8e9e8aac86 in rosbag2_transport::Rosbag2Transport::record (this=0x7f8e975ad090, storage_options=..., record_options=...)
    at /home/jenkins-agent/workspace/packaging_linux/ws/src/ros2/rosbag2/rosbag2_transport/src/rosbag2_transport/rosbag2_transport.cpp:70
#18 0x000055df8f8bb427 in RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::{lambda()#1}::operator()() const ()
#19 0x000055df8f8cf69d in void std::__invoke_impl<void, RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::{lambda()#1}>(std::__invoke_other, RecordIntegrationTestFixture::start_recording(rosbag2_transport::RecordOptions const&)::{lambda()#1}&&) ()
[... cut ...]
    at /home/jenkins-agent/workspace/packaging_linux/ws/src/eProsima/Fast-RTPS/include/fastrtps/rtps/reader/../common/Guid.h:108
  13   Thread 0x7f8e8dfad700 (LWP 25323) 0x00007f8e9def4567 in __libc_recvmsg (fd=fd@entry=18, msg=msg@entry=0x7f8e8dfac240, flags=flags@entry=0) at ../sysdeps/unix/sysv/linux/recvmsg.c:28
  14   Thread 0x7f8e8cfab700 (LWP 25325) clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:78
  15   Thread 0x7f8e6ffff700 (LWP 25326) clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:78
  16   Thread 0x7f8e6f7fe700 (LWP 25327) clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:78
```

pbaughman, 2019-06-28:
I can get it to segfault pretty easily by adding some CPU stress at the same time the test is running.  Run the test in a loop in one terminal, and then in a second terminal run

>stress -c 12

In another terminal.  The test will segfault within a minute or two

emersonknapp, 2020-01-29:
Putting on backlog to see if we can reproduce this now:
suggestions, run with ASAN/TSAN

rotu, 2020-03-24:
Seems to be the same as https://github.com/ros2/rmw_cyclonedds/issues/118

Karsten1987, 2020-03-31:
Looking at the strack trace, I consider this ticket being the same as #329 
I think this is being addressed in #338 and thus I am going to close this ticket.

@pbaughman please feel free to re-open if you still encounter this issue.

closed 2020-03-31 by Karsten1987

rotu, 2020-04-01:
I agree the segfault is fixed. It looks like the original hang might still be out there.