ros2/rosbag2 #139: how can i use bag file from ROS1 with rosbag2

sunhuchang, 2019-07-09:
in ros2 dashing,I run rosbag2 but failed:
# ros2 bag info ~/Downloads/cartographer_paper_deutsches_museum.bag 
[ERROR] [rosbag2_storage]: Failed to load metadata: Exception on parsing info file: bad file
[ERROR] [rosbag2_storage]: Could not open '/home/orion-jetsonnano/Downloads/cartographer_paper_deutsches_museum.bag' with 'sqlite3'. Error: Failed to read from bag '/home/orion-jetsonnano/Downloads/cartographer_paper_deutsches_museum.bag': No metadata found.
[ERROR] [rosbag2_storage]: Could not load/open plugin with storage id 'sqlite3'.
[ERROR] [rosbag2_transport]: Could not read metadata for /home/orion-jetsonnano/Downloads/cartographer_paper_deutsches_museum.bag. Please specify the path to the folder containing an existing 'metadata.yaml' file or provide correct storage id if metadata file doesn't exist (see help).

sunhuchang, 2019-07-09:
the bag file i used  is download from cartographer, I dont know how to find the corespondence yaml.

Karsten1987, 2019-07-17:
You're downloading a ROS1 bag file, but you try to replay it with sqlite3 which is the default format for ROS2. 
So you want to replay your bag with 
```
ros2 bag play -s rosbag_v2 /home/orion-jetsonnano/Downloads/cartographer_paper_deutsches_museum.bag
```

rsasaki0109, 2019-07-22:
Hello,@Karsten1987.
I am faced with the same problem of this issue.(I could record and play 'sqlite3'file.)
Following your advice, I tried this.
```
ros2 bag play -s rosbag_v2 infant_outdoor.bag 
```
but ,an error occured.
```
[ERROR] [rosbag2_storage]: Could not load/open plugin with storage id 'rosbag_v2'.
[ERROR] [rosbag2_transport]: Failed to play: No storage could be initialized. Abort

```
Help me plz...

 - my enivironment 
ubuntu 18.04
ros dashing
 - my rosbag information
https://drive.google.com/file/d/1VEy_iJZKEGcNDDKsK-YrqgKxwy6qsric/view
```
rsasaik@ubuntu:~/ros2_basics$ rosbag info infant_outdoor.bag 
path:        infant_outdoor.bag
version:     2.0
duration:    4:33s (273s)
start:       Nov 21 2018 18:42:37.73 (1542793357.73)
end:         Nov 21 2018 18:47:11.39 (1542793631.39)
size:        591.3 MB
messages:    43751
compression: none [777/777 chunks]
types:       nav_msgs/Odometry          [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/Imu            [6a62c6daae103f4ff57a132d6f95cec2]
             velodyne_msgs/VelodyneScan [50804fc9533a0e579e6322c04ae70566]
topics:      /imu/data           27367 msgs    : sensor_msgs/Imu           
             /tinypower/odom     10946 msgs    : nav_msgs/Odometry         
             /velodyne_packets    5438 msgs    : velodyne_msgs/VelodyneScan
```

rsasaki0109, 2019-07-22:
Hello,@Karsten1987.
I am faced with the same problem of this issue.(I could record and play 'sqlite3'file.)
Following your advice, I tried this.
```
ros2 bag play -s rosbag_v2 infant_outdoor.bag 
```
but,an error occured.
```
[ERROR] [rosbag2_storage]: Could not load/open plugin with storage id 'rosbag_v2'.
[ERROR] [rosbag2_transport]: Failed to play: No storage could be initialized. Abort

```
Help me plz...

 - my enivironment 
ubuntu 18.04
ros dashing
 - my rosbag information
https://drive.google.com/file/d/1VEy_iJZKEGcNDDKsK-YrqgKxwy6qsric/view
```
sasaki@ubuntu:~/ros2_basics$ rosbag info infant_outdoor.bag 
path:        infant_outdoor.bag
version:     2.0
duration:    4:33s (273s)
start:       Nov 21 2018 18:42:37.73 (1542793357.73)
end:         Nov 21 2018 18:47:11.39 (1542793631.39)
size:        591.3 MB
messages:    43751
compression: none [777/777 chunks]
types:       nav_msgs/Odometry          [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/Imu            [6a62c6daae103f4ff57a132d6f95cec2]
             velodyne_msgs/VelodyneScan [50804fc9533a0e579e6322c04ae70566]
topics:      /imu/data           27367 msgs    : sensor_msgs/Imu           
             /tinypower/odom     10946 msgs    : nav_msgs/Odometry         
             /velodyne_packets    5438 msgs    : velodyne_msgs/VelodyneScan
```

Karsten1987, 2019-07-22:
are you having the package installed? The `rosbag_v2` is a plugin which has to be installed separately.

You can either install it via 
```
sudo apt install ros-dashing-rosbag2-bag-v2-plugins
```
or clone and compile the package yourself:
```
git clone https://github.com/ros2/rosbag2_bag_v2
```

AAlon, 2019-07-24:
I'm having the same issue:
```
[ERROR] [rosbag2_storage]: Could not load/open plugin with storage id 'bag_v2'.
[ERROR] [rosbag2_transport]: Failed to play: No storage could be initialized. Abort
```

Karsten1987, 2019-07-24:
@AAlon the plugin identifier is called `rosbag_v2`. This has to be exact. `bag_v2` does not exist.

rsasaki0109, 2019-07-24:
@Karsten1987
Following your advice and the issue #94, 
I succeeded to run rosbag2.
Thank you!

AAlon, 2019-07-29:
> @AAlon the plugin identifier is called `rosbag_v2`. This has to be exact. `bag_v2` does not exist.

I tried `rosbag_v2` before that, same error:
```
 ros2 bag play -s rosbag_v2 ~/all_hands_raw.bag
[ERROR] [rosbag2_storage]: Could not load/open plugin with storage id 'rosbag_v2'.
[ERROR] [rosbag2_transport]: Failed to play: No storage could be initialized. Abort
```

When I use the ROS2-D from system installation (after `sudo apt install ros-dashing-rosbag2-bag-v2-plugins`), I'm getting:
```
[ERROR] [rosbag2_storage]: Unable to load instance of read write interface: Failed to load library /opt/ros/dashing/lib/librosbag2_bag_v2_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libroscpp_serialization.so: cannot open shared object file: No such file or directory)
[ERROR] [rosbag2_storage]: Could not load/open plugin with storage id 'rosbag_v2'.
[ERROR] [rosbag2_transport]: Failed to play: No storage could be initialized. Abort
```

AAlon, 2019-07-29:
As suspected I was able to get it working by having ROS1 as well:
```
source /opt/ros/melodic/setup.bash
source /opt/ros/dashing/setup.bash
ros2 bag play -s rosbag_v2 <file>
```
It would be great if you could update the documentation to mention that. Thanks!

Karsten1987, 2019-08-05:
@AAlon I totally agree. There has to be more documentation. I've ticketed it on the rosbag2_bag_v2 repository: https://github.com/ros2/rosbag2_bag_v2/issues/5

Closing this issue as I believe it's solved.

closed 2019-08-05 by Karsten1987

ShuiYunXi, 2022-05-05:
[ERROR] [1651739381.334239687] [rosbag2_storage]: Unable to load instance of read write interface: Failed to load library /opt/ros/foxy/lib/librosbag2_bag_v2_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library LoadLibrary error: libroscpp_serialization.so: cannot open shared object file: No such file or directory, at /tmp/binarydeb/ros-foxy-rcutils-1.1.3/src/shared_library.c:84
[ERROR] [1651739381.334288047] [rosbag2_storage]: Could not load/open plugin with storage id 'rosbag_v2'.
[ERROR] [1651739381.334328926] [rosbag2_transport]: Failed to play: No storage could be initialized. Abort
----------------
hello, I'm using foxy to run ros1 bag,but when I use the ros-foxy-rosbag2-bag-v2-plugins and it's report an error!Maybe it not support on foxy?