ros2/rosbag2 #142: lsan linking issue

christianrauch, 2019-07-24:
When building on Raspbian armv6l, I get a linking issue for `test_ros2_message`:
```
/usr/bin/ld: cannot find liblsan_preinit.o: No such file or directory
/usr/bin/ld: cannot find -llsan
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_ros2_message.dir/build.make:156: test_ros2_message] Error 1
make[1]: *** [CMakeFiles/Makefile2:189: CMakeFiles/test_ros2_message.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

Karsten1987, 2019-08-05:
I am currently not able to reproduce this. Is there an easy way (docker or similar) for me to test this?

christianrauch, 2019-08-05:
I only encountered the issue on a Raspberry Pi 1 with Raspbian Buster. I am not sure if it is possible to reproduce the issue without the hardware. I can try to create a minimal workspace for Raspbian Buster, that can be used with the official Raspbian SD card image. But I assume you will need to run it on real hardware.

Karsten1987, 2019-08-05:
If you can by any chance come up for a possible solution which solves the problem for your setup, I am happy to run CI on it to see if it's compatible with non ARMV6 hardware and potentially get it merged upstream.

inochi108, 2019-08-12:
I got the same error.  My configuration is a Raspberry Pi 3B with Ubuntu Mate 18.04.02 Bionic for a Raspberry Pi ARMv7 32-bit system.  Building ROS2 dashing (https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/) from source with command "sudo colcon build --symlink-install --executor sequential".  I skipped the optional steps and building the ros1 bridge.

Image:  https://ubuntu-mate.org/raspberry-pi/ubuntu-mate-18.04.2-beta1-desktop-armhf+raspi-ext4.img.xz

kbreitton, 2019-08-20:
I had the same issue. liblsan is apparently not available for 32bit systems 
https://github.com/google/sanitizers/issues/794

Karsten1987, 2019-08-20:
Okay, I think I understand. It makes sense to avoid the sanitizers then when they are not available on x86-64. 
Can somebody of you please verify if adding the cmake option `DISABLE_SANITIZERS` work here?
https://github.com/ros2/rosbag2/blob/master/rosbag2/CMakeLists.txt#L18

So essentially run `colcon build --symlink-install --cmake-args -DDISABLE_SANITIZERS=ON`. 

If so, I believe the easiest fix is to exclude 32bit systems for this test.

kbreitton, 2019-08-20:
Yup, running with DISABLE_SANITIZERS=ON worked!

Karsten1987, 2019-08-21:
glad it worked. I've open #149 which disables the test for non 64bit machines. As I am not having another machine around, could you please try that patch as well?

CMake makes it rather hard to find out reliably which architecture is used (especially when cross compiling) so I enabled the test only when x64 is the host system. 
I am not completely convinced that the PR is the best solution and if passing the option to disable sanitizers works fine for you, we can also just take this as the recommended solution.

Karsten1987, 2019-09-19:
fixed by #142

closed 2019-09-19 by Karsten1987

christianrauch, 2019-09-19:
@Karsten1987 I guess you meant #149.