ros2/rosbag2 #152: Compression Options

LuAPi, 2019-08-30:
ROS bag 1 supported compression. Are there plans to allow ROS bag 2 to compress messages too?

I found the compression option in ROS bag 1 suboptimal for our uses. The main issue was that the whole process was inherently serial and didn't allow for parallel compression. I looked into how one could implement asynchronous parallel compression but found too much of ROS bag 1 was setup for serial operation for it to be possible to implement without a major rewrite.

I had a brief look into the code and I think it might be reasonable to include an integer field in https://github.com/ros2/rosbag2/blob/master/rosbag2_storage/include/rosbag2_storage/serialized_bag_message.hpp that is a `COMPRESSION_TYPE_ID` value which references a table of compression types.  Those ID numbers should reference another table which maps `COMPRESSION_TYPE_ID` to some identifying string. The default `COMPRESSION_TYPE` `0` would be `NONE`.

Some desirable features:
- Compression is applied to each message asynchronously in parallel by a thread from a thread pool (with command line option to specify number of compression threads).
- Ability to specify compression type by channel / message type.
- Options for different compression algorithms - allows trading off size & speed or using algorithms designed for specific types of data (like image compression on an image topic).
- Options for what to do if messages are coming in faster than they can be compressed - drop messages or store uncompressed versions.
- Ability to compress existing rosbag2 files
  - When doing so gain options of using compression methods that can build a dictionary from the messages to be compressed and then compress the messages using that dictionary which is also stored in the bag file.
  - In place [de]compression would be done by writing the messages to a new rosbag2 file, verifying the new file is valid and then replacing the original file with the new one. Not the way rosbag does it by renaming the original file to have .orig in its name and then leaving it like that if something goes wrong. Doing it the rosbag1 way makes writing scripts to process rosbag files troublesome.

piraka9011, 2019-12-12:
> Ability to compress existing rosbag2 files 

#220 and following should be addressing this

LuAPi, 2019-12-13:
I've had a look at that and the related PRs (both merged and unmerged).

I have a few concerns.
The Interfaces define two methods of [de]compressing things. One for handling whole files the second for individual messages.
Looking at the implementation of the one for handling whole files:
- [minor concern] It'd be better if you could optionally specify the output file uri instead of letting the code generate one.
- [major concern] It looks like the implementation actually just applies compression to the entire file. That means it ceases to be a valid bag file and doesn't get you anything that you couldn't achieve by using any other file compression utility. I'd have expected the behaviour to be that it opens the input bag file, creates a new output bag file and then iterates through all the messages in the input, applies compression to them [using a specified number of threads] and writes them to the output bag file.
- The current implementation reads the whole file into memory, applies compression and then writes the data to an output file. The bag files from our robots can be multiple hundred gigabytes in size which is much much larger than the available ram on the robots, and probably would exceed the ram on our servers given the need to hold both uncompressed and compressed versions of the data in ram at once. Zstd has a streaming API which would be better for that use case, but I'm guessing that the current code is mostly a placeholder.

On the compressing individual messages side of things:
- It doesn't look like it from the interface, but there was talk in the comments about compressing individual messages in place. I am a little concerned about robustness to failures - I wasn't sure how much thought had been given to what happens if a failure occurs in the middle of the compression operation. I'm not sufficiently familiar with this codebase to judge that. Preserving the integrity of our log files is  big concern for our organisation so operations that leave data files in a broken state if they fail part way through are a big no no for us.
- It seems that the type of compressions is associated with the whole file rather than individual messages like my suggestion. It's not a major issue since if we find we would like to be able to for example apply draco to pointclouds, png to images and zstd to everything else we could write our own compressor implementation, though the current interface doesn't provide a way to do that to only some specific channels.
- It might be present elsewhere but I didn't see much discussion about concurrency. I'm not familiar enough with this codebase so I'm not sure where these [de]compress individual messages get used so I don't know if it's the case or not; but it would be desirable to be able to specify how many threads get used to compress messages when recording topics.

There was mention of a design document in one of the threads but no link to it. It's also not clear how set in stone ROS2 design decisions are right now so it's not clear what things I could contribute to without the risk of a PR being rejected because it would break compatibility with older ROS2 versions [such as changing the bag file format or a standard message type].

piraka9011, 2019-12-13:
> It looks like the implementation actually just applies compression to the entire file. 

The current implementation is only on a macro (file) level. The `compress_serialized_bag_message` which is for message compression would perform your described workflow (single threaded though). 

If my understanding is correct, rosbag2 is currently running on a single threaded executor meaning operations could be performed asynchronously. @Karsten1987 might have a better understanding on how to parallelize processes.

> The current implementation reads the whole file into memory... Zstd has a streaming API...

We are aware of that and using streaming would be the next step. A proposed approach is sanitizing the input to enforce splitting by a certain file size when compressing to avoid loading large files into memory. If you would like to work on the streaming API, your contribution would be appreciated :smiley: 

> I wasn't sure how much thought had been given to what happens if a failure occurs in the middle of the compression operation.

This is also a valid concern. One suggestion was to simply not compress the message and log a warning as opposed to dropping messages.

> It's also not clear how set in stone ROS2 design decisions...

I don't see what you can't open a PR against the [Rosbag2 design article](https://github.com/ros2/design/blob/f69fbbd11848e3dd6866b71a158a1902e31e92f1/articles/rosbags.md) with suggestions or an issue (such as this one). Also, there's no harm in opening a PR against the [ros2/design](https://github.com/ros2/design) repo in general for any suggestions/comments.

emersonknapp, 2021-02-19:
I'm going to close this as I think all of the requested features have been implemented - file compression, message compression, zstd streaming API, multithreaded compression.

It's hard to use very generic tickets that can live forever, so if any specific feature asks remain, please feel free to reopen those as targeted tickets.

closed 2021-02-19 by emersonknapp