ros2/rosbag2 #153: Bag recorded on 32bit ARM is not readable

Stapelzeiger, 2019-09-03:
A bag recorded on a 32bit armhf Linux target can't be played. Trying to play it on the same armhf target gives the following error:
```
[ERROR] [rosbag2_storage]: Failed to load metadata: Exception on parsing info file: yaml-cpp: error at line 7, column 18: bad conversion
[ERROR] [rosbag2_storage]: Could not open 'rosbag2_2019_09_02-22_37_16' with 'sqlite3'. Error: Failed to read from bag 'rosbag2_2019_09_02-22_37_16': No metadata found.
[ERROR] [rosbag2_storage]: Could not load/open plugin with storage id 'sqlite3'.
[ERROR] [rosbag2_transport]: Failed to play: No storage could be initialized. Abort
```

Opening the bag on macOS amd64 has the following error:
```
[ERROR] [rosbag2_transport]: Failed to play: SQL error when preparing statement 'SELECT name, type, serialization_format, COUNT(messages.id), MIN(messages.timestamp), MAX(messages.timestamp) FROM messages JOIN topics on topics.id = messages.topic_id GROUP BY topics.name;' with return code: 14
```

Bags recorded on macOS can be played back on macOS, but not on armhf/linux (they fail the same way, when parsing the yaml).
This is using Dashing patch release 2 on both platforms.

Karsten1987, 2019-09-04:
Looking into the return codes of SQLite3, I believe [error code 14](https://sqlite.org/rescode.html#cantopen) states to be unable to open the database file. 

I believe this makes sense as SQLite3 stores its data in binary form and thus I could see that it's architecture dependent. I believe thus a bag recorded on x64 machines can only be replayed on x64 machines.

I have to look into why a bag file recorded on 32bit armhf can't be opened. I don't have a 32bit machine around, but could you tell me a bit more about your setup and how I could easily reproduce it?

sloretz, 2019-09-04:
> I believe this makes sense as SQLite3 stores its data in binary form and thus I could see that it's architecture dependent. I believe thus a bag recorded on x64 machines can only be replayed on x64 machines.

@Karsten1987 https://www.sqlite.org/onefile.html

> A database in SQLite is a single disk file¹. Furthermore, the file format is cross-platform. A database that is created on one machine can be copied and used on a different machine with a different architecture. SQLite databases are portable across 32-bit and 64-bit machines and between big-endian and little-endian architectures.

Karsten1987, 2019-09-04:
Thanks @sloretz that's great insights! I'll have to look into this then once I am able to set up some 32bit machine.

Stapelzeiger, 2019-09-04:
> I have to look into why a bag file recorded on 32bit armhf can't be opened. I don't have a 32bit machine around, but could you tell me a bit more about your setup and how I could easily reproduce it?

I was running the talker demo and recorded the bag with `ros2 bag record -a`. The platform I'm using is a NanoPi duo 2 with Ubuntu Bionic, but I guess a Raspberry should be the same. The ros2 binaries are downloaded from the github releases.
Then I copy the bag to my mac and try to open it with `ros2 bag play <bagname>`.

I can send you the bag file if that helps.

Stapelzeiger, 2019-09-04:
Actually, I just observed something interesting: The bag, which is recorded on the ARM and doesn't open on macOS, opens fine in my Linux amd64 VM. On top of that opening the bag in the VM generates two files in the bag (.db3-shm and .db3-wal). With those files present, that the bag opens fine in macOS.
I checked the .db3/.yaml are not modified, only the two other files mentioned are added.

emersonknapp, 2021-08-06:
Closing as stale - Dashing has reached EOL, and no supported ROS 2 distros have Tier 1 ARM32 support. Would accept investigations/bugfixes, if desired, however.

closed 2021-08-06 by emersonknapp