ros2/rosbag2 #154: Problem when i used bag file from ROS1 with 'ros2 bag play -s rosbag_v2'

EgalYue, 2019-09-03:
Hi,
I want to **play bag file from ROS1 with ROS2 bag**, and i installed the **rosbag2_bag_v2** plugin.
when i used `ros2 bag info -s rosbag_v2 testbag.bag`, everything is fine.
testbag.bag is my test bag file.
[image]

But when i used `ros2 bag play -s rosbag_v2 testbag.bag` to **play** the bag, i got nothing!

[image]
-------------------------------------------------------
And i did another test, i downloaded the bag file(rgbd_dataset_freiburg1_rpy.bag) form TUM dataset.
`ros2 bag info -s rosbag_v2 ~/rgbd_dataset_freiburg1_rpy.bag` is ok!
[image]

But `ros2 bag play -s rosbag_v2 ~/rgbd_dataset_freiburg1_rpy.bag` just printed like following:
 
[image]

I dont know, why is this so?
Please help me!
Thank you!

------------------------------------------
I run this in a **docker** container:
ros2: dashing
ros1: melodic

Karsten1987, 2019-09-03:
two little remarks up front:

* The `play` command for `ros2 bag` is pretty silent while playing. Meaning you won't see anything printed on the terminal even though the bag file is currently being replayed. 
* The `rosbag_v2` plugins requires the `ros1_bridge` to be compiled with all available message types. The bridge which is shipped with the Debians\*, doesn't know about any custom ROS messages and thus has to be built from source. 

Now, in order to help you get forward, can you please tell me about which setup you are using? Are you compiling ROS2 dashing and the ros1 bridge from source or are you using the provided debs? 
Given that `ros2 bag info` works, that indicates that the plugin was installed and loaded correctly. The rosbag2 package provides [test bags](https://github.com/ros2/rosbag2_bag_v2/blob/master/rosbag2_bag_v2_plugins/resources/test_bag_end_to_end.bag)  which are used within the unit tests. Can you verify that these work for you?
I could verify that `ros2 topic echo /string_topic std_msgs/msg/String` outputs the correct data when replying the bag file.

---
*https://github.com/ros2/ros1_bridge#bridge-communication-between-ros-1-and-ros-2
> The bridge provided with the prebuilt ROS 2 binaries includes support for common ROS interfaces (messages/services), such as the interface packages listed in the ros2/common_interfaces repository and tf2_msgs

EgalYue, 2019-09-04:
Hi Karsten1987,
Thank you for your answer. 
Im using the provided debs (ROS2 and ros1 bridge). `apt install ros-dashing-ros1-bridge`

I got it, ("The play command for ros2 bag is pretty silent while playing"). I got the message through rviz2! it works fine!

For the `tf/tfMessage`, i need to make a setup from the following page, right?
https://github.com/ros2/ros1_bridge/blob/master/doc/index.rst

Karsten1987, 2019-09-04:
glad it worked out for you so far. I am going ahead and close this issue then as I believe the remaining work will be related to setting up the bridge for custom messages. Feel free to re-open it though in case you encounter further problems with rosbags.

> For the tf/tfMessage, i need to make a setup from the following page, right?
https://github.com/ros2/ros1_bridge/blob/master/doc/index.rst

That seems definitely the right entry point for you.

closed 2019-09-04 by Karsten1987

EgalYue, 2019-09-09:
Hi,
I met one problem, when I want to get `/tf` message from my rosbag or from TUM dataset rosbag?

I installed and built `ros1_bridge` from source.
I run this in a docker container:
ros2: dashing (using the provided debs)
ros1: melodic (using the provided debs)

**The following are my steps:**
Step 1: Open shell 1:
```
. /opt/ros/melodic/setup.bash
. /opt/ros/dashing/setup.bash
. ./bridge_ws/install/setup.bash
ros2 bag info -s rosbag_v2 rgbd_dataset_freiburg1_rpy.bag
ros2 bag play -s rosbag_v2 rgbd_dataset_freiburg1_rpy.bag

```
[image]

 And the type of  `/tf` topic is `tf2_msgs/msg/TFMessage`, it looks ok.

Step 2: Open shell 2:
```
. /opt/ros/melodic/setup.bash
. /opt/ros/dashing/setup.bash
. ./bridge_ws/install/setup.bash
ros2 topic list
ros2 topic echo /tf
```
[image]

**topic `/tf` showed nothing!**

Please help !
Thank you!