ros2/rosbag2 #155: Provide way to write Log messages as plaintext to rosbags

emersonknapp, 2019-09-06:
Issue Type: Feature request.

## Motivation
For users recording /rosout logs to a rosbag, there isn't a good way to for a user to explore these logs with SQL queries since they're not human-readable.

## Description
Implement and provide a default converter plugin to write messages as plaintext so they can be explored by a human.

## Requirements
* Acceptable to limit the plugin to handle rcl_interfaces/Log message type only
* Acceptable to be a Serializer only, does not need to deserialize since the intended consumer is a person

emersonknapp, 2019-09-06:
I am currently working on this - issue opened to track work

dirk-thomas, 2019-09-06:
Why should the recording convert the existing log message (https://github.com/ros2/rcl_interfaces/blob/a7ea280423e64f38d910e318d73a55bc22e71d42/rcl_interfaces/msg/Log.msg) into a different format and then also write it to the bag (beside the already recorded message)? I don't think that makes any sense.

Maybe you could write e.g. a `bag2csv` script which reads a bag file and extracts all log messages into whatever format you would like to have.

emersonknapp, 2019-09-06:
We are trying to have human-readable output at recording time, rather than an after-the-fact conversion of bags into a separate file. For example we could provide a CSV storage plugin to write the bag to CSV instead of a sqlite3 DB, again rather than after-the-fact conversion. 

But we believe this simple plugin is the smallest possible thing that we can build and get into Dashing, to unblock the needed end user workflow of just being able to read stored logs (since ROS2 does not yet have the ability to write logs to files).

dirk-thomas, 2019-09-06:
> We are trying to have human-readable output at recording time

Why not write a ~ twenty lines Python script which subscribes to the `/rosout` topic and writes the data into whatever file format you need? Why use `rosbag` at all for this task - what is the benefit?

emersonknapp, 2019-09-06:
That is a nice workaround, too.

But I do see value in utilizing this common recording tool to have the ability to output something in a format that can be explored by someone in-place, rather than only being useful for playback. 

The benefit, as I see it, is that there already exists here a subscription/autodiscovery mechanism (so we can use different topics automatically), and an architecture to support a variety of input and output formats (so we could use this text-out formatter to go to other storage backends), such that this is not a one-off thing but a feature that extends the combinatorial usefulness of rosbag2.

Perhaps you think this ought to be re-scoped a little? Or that bags should never store plaintext?

dirk-thomas, 2019-09-06:
I really don't mind. If you think it fits your use case please go ahead with your idea.

I was only commenting with my perspective since the described feature seems far off from what rosbag is targeted to do. Also I am not a developer of this package so I don't have an opinion what makes sense or doesn't to be added to it.

emersonknapp, 2019-09-06:
I appreciate the input!

Karsten1987, 2019-09-09:
I can understand the reasoning behind wanting to have message types serialized differently inside a rosbag file.
I would question the mentioned requirements though and would rather envision introducing a new argument to the command line tool which lets you specify a list of topics to be applied to this converter. This is basically to avoid hardcoding the converter plugin to only deal with `rcl_interfaces/Log` messages, e.g. it would make sense to me also to write `std_msgs/String` differently.

emersonknapp, 2019-09-09:
I think that reframing makes sense - from the current state, does current lack of multi-serialization-format support add a significant complexity to implementation (https://github.com/ros2/rosbag2/issues/52)?

Also after considering @dirk-thomas input more, for our immediate need we will write a simple node to subscribe and dump to text file, as suggested.

However, I think this feature is still probably valuable in the big picture. I agree that std_msgs/String should also be able to be plaintext logged. The single-direction thought was just for simplicity of implementation - since we may have more time now, I think that both ways makes more sense.

Based on that, here's an idea for a rewrite of this feature:
* CSV serialization format plugin
  * bidirectional to support playback
  * fields can be gathered from the topic type
  * we may need to consider "snapshot" the message definition in the metadata because it may change over time. this problem already exists though, I assume? how do we think about this?

emersonknapp, 2020-01-29:
Logging to file is working more reliably, this feature is no longer needed as a workaround, no plan to add it as a default plugin

closed 2020-01-29 by emersonknapp