ros2/rosbag2 #164: segfault on deserialize

eric1221bday, 2019-09-23:
I've been trying to read rosbag2's via the `SequentialReader` API. Using the `SerializationFormatConverterFactory` to load the "cdr" deserializer like such.
```cpp
auto converter_factory = std::make_unique<rosbag2::SerializationFormatConverterFactory>();
deserializer = converter_factory->load_deserializer(rmw_get_serialization_format());
```
I then am using a pattern I saw in this repo to deserialize the message and obtain the ROS msg.
```cpp
void MapBuilder::ReadBag(const std::string& bag_path) {
  reader->open({bag_path, "sqlite3"}, {"", rmw_get_serialization_format()});

  std::unordered_map<std::string, rosbag2::ConverterTypeSupport> typesupport_map;

  for (auto const& metadata : reader->get_all_topics_and_types()) {
    auto output_metadata = metadata;
    output_metadata.serialization_format = rmw_get_serialization_format();
    rosbag2::ConverterTypeSupport type_support;
    type_support.rmw_type_support = rosbag2::get_typesupport(metadata.type, "rosidl_typesupport_cpp");
    type_support.introspection_type_support =
        rosbag2::get_typesupport(metadata.type, "rosidl_typesupport_introspection_cpp");
    typesupport_map.emplace(metadata.name, type_support);
    std::cout << metadata.name << ", " << metadata.type << std::endl;
  }

  while (reader->has_next()) {
    auto serialized_msg = reader->read_next();

    auto ts = typesupport_map.at(serialized_msg->topic_name).rmw_type_support;
    auto introspection_ts = typesupport_map.at(serialized_msg->topic_name).introspection_type_support;
    std::shared_ptr<rosbag2_introspection_message_t> allocated_ros_message =
        rosbag2::allocate_introspection_message(introspection_ts, &allocator);
    deserializer->deserialize(serialized_msg, ts, allocated_ros_message);
  }
}
```

Unfortunately the deserialize call simply segfaults, and since this is a raw plugin load gdb just segfaults more or less on entry to the function.

I looked through the current code base and it seems as if this particular case has not actually been tested? The CdrConverter plugin test was compiled directly with the plugin, and none of the other functionalities implemented so far seems to need to deserialize a cdr.

I'm a bit stumped on whether I am doing something wrong or if there's a problem with the rosbag code itself. Any help would be appreciated.

Karsten1987, 2019-11-11:
A question up front: Are you trying to read a bag file which was recorded with a different serialization format than cdr? Because, really, you'd only need the converter plugins for that case. If not, you can simply read the cdr serialized ros messages and use [rmw_deserialize](https://github.com/ros2/rmw/blob/master/rmw/include/rmw/rmw.h#L536-L558) to obtain a ros message instance.

> I looked through the current code base and it seems as if this particular case has not actually been tested?

There is a test for the cdr converter plugin: https://github.com/ros2/rosbag2/blob/master/rosbag2_converter_default_plugins/test/rosbag2_converter_default_plugins/cdr/test_cdr_converter.cpp

I assume that you have all message types of the rosbag in question locally installed on your machine? You'd need all messages being installed in order to call the respective typesupport for it.

While you are already debugging with GDB, can you verify that the function pointers for deserialization are set correctly?
Essentially these lines here:
https://github.com/ros2/rosbag2/blob/master/rosbag2_converter_default_plugins/src/rosbag2_converter_default_plugins/cdr/cdr_converter.cpp#L107-L108 
`deserialize_fcn_` is essentially a pointer to `rmw_deserialize` within `rmw_fastrtps_cpp`.

eric1221bday, 2019-12-05:
@Karsten1987 sorry for the late response. I am indeed just reading the bag with cdr serialization format. I have actually looked at the cdr_converter code before, and the different between that test and my use case is the plugin architecture. While the test explicitly links the converter in at build time. In normal use case it would be directly loaded via the plugin interface. 

Is there a way for me to check what the function address of rmw_deserialize is in gdb?

emersonknapp, 2022-10-04:
I'm going to close this issue as out-of-date given the availability now of the `rclcpp::Serialization` classes. Please reopen if there is a specific and remaining issue that isn't addressed by this.

closed 2022-10-04 by emersonknapp