ros2/rosbag2 #174: need for metapackage

Karsten1987, 2019-10-09:
I think there's the need for a sort of a metapackage. This package can then be used conveniently for triggering CI but also be part of the ros2-variants. 

I am thinking about `rosbag2_common_extensions`. Any idea for names?

wjwwood, 2019-10-16:
What would this "meta" package depend on that `rosbag2` does not?

Karsten1987, 2019-10-17:
`rosbag2_transport` and `ros2bag` (the cli) are on top of `rosbag2`.

I like @wjwwood's idea of renaming `rosbag2` to something more precise as `rosbag2_core` or `rosbag2_cpp` and introduce `rosbag2` as the meta package. That'd allow for the debian package to be `ros-<version>-rosbag2` which makes a lot of sense to me.

wjwwood, 2019-10-17:
The only downside is the cpp namespace should really match the package name. So if you want to keep using the rosbag2 namespace I’m cpp it really ought to be in a package of the same name.

Karsten1987, 2019-12-10:
I am proposing to proceed as follows:

* duplicate `rosbag2` package and rename to `rosbag2_cpp`. 
* adjust code within `rosbag2_cpp` to new name .. include headers, namespace, etc.
* adjust `rosbag2_transport` to use `rosbag2_cpp`.
* cleanup `rosbag2` to only contain a basically empty `CMakeLists.txt` as well as a `package.xml` with all the dependencies listed. 
* add `rosbag2` as dependency to `ros2-variants` - either `ros_core` or `ros_base`

opinions?

wjwwood, 2019-12-10:
Sounds good to me, I like `rosbag2_cpp` as the API name too.

PR #241 (2020-01-06): Make rosbag2 a metapackage
fixes #174 

As discussed in the top level ticket, this PR renames the existing `rosbag2` package to `rosbag2_cpp` and introduces `rosbag2` as a simple metapackage with only listing runtime dependencies.
The renaming also entails changing the namespace accordingly, which makes that PR pretty brutal.
closed 2020-01-06 by Karsten1987