ros2/rosbag2 #179: rosbag2_transport cannot be found on Windows

hidmic, 2019-10-17:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - From sources
- Version or commit hash:
  -  
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

When using the `ros2` command, lots of errors come up:

```
Failed to load entry point 'info': DLL load failed: The specified module could not be found.
The C extension 'c:\j\workspace\ci_windows\ws\install\lib\site-packages\rosbag2_transport\_rosbag2_transport_py.pyd' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'play': DLL load failed: The specified module could not be found.
The C extension 'c:\j\workspace\ci_windows\ws\install\lib\site-packages\rosbag2_transport\_rosbag2_transport_py.pyd' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
Failed to load entry point 'record': DLL load failed: The specified module could not be found.
The C extension 'c:\j\workspace\ci_windows\ws\install\lib\site-packages\rosbag2_transport\_rosbag2_transport_py.pyd' failed to be imported while being present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/#import-failing-even-with-library-present-on-the-system' for possible solutions
``` 

Seen on https://ci.ros2.org/job/ci_windows/8300/testReport/junit/ros2topic.test/test_cli/test_cli/ first.

#### Expected behavior
I'd expect no errors.

#### Actual behavior
Errors arise

hidmic, 2019-10-18:
The problem seems to stem from C extensions being imported at the module level, like [this one](https://github.com/ros2/rosbag2/blob/dc2b161f7f019d595edcac926f227a088ea175e4/rosbag2_transport/rosbag2_transport/__init__.py#L32). That does not play along `colcon` workspaces in Windows. CC @Karsten1987.

closed 2019-10-21 by hidmic

Karsten1987, 2019-10-21:
thanks @hidmic for fixing this!

PR #190 (2019-10-21): Import rosbag2_transport Python module on demand
Closes #179. This pull request ensure `ros2bag` imports `rosbag2_transport` on demand, as opposed to doing so at the module level. 

I purposefully didn't rehash the `rosbag2_transport` module to enforce delayed imports as there's been some discussion and work around removing this limitation, and hopefully we might at some point in the future (see https://github.com/ros2/rclpy/pull/417, https://github.com/ros2/rclpy/pull/420 and https://github.com/ros2/rclpy/pull/422).