ros2/rosbag2 #193: Be able to record custom message types

emersonknapp, 2019-10-22:
Currently, rosbag2 can only record message types that it is compiled against (so, the standard ros2 core messages only). We should be able to record any messages in the system, like ROSbag1 did, so that custom applications can be recorded and played back.

Some speculation for info:
It sounds like the `rmw_*_dynamic_cpp` APIs are what can enable this feature.

emersonknapp, 2019-10-22:
@Karsten1987 can you confirm the above? I'm not 100% certain about it but want to track if true.

Karsten1987, 2019-10-22:
just for clarification, rosbag2 is not compiled against any message libraries. However, what is true is that you need to have the message library on your system in order to load the typesupport handle. This is not so much a rosbag2 problem as more of a general ROS2 problem as you currently can't publish/subscribe to messages you don't have locally installed on your system.

emersonknapp, 2019-10-22:
Okay, that is a better situation than we thought. So what I'm hearing is that a publisher on your ROS2 network may publish a type you do not have installed on the same computer as `rosbag2`, and in that case you will not be able to understand it.

That being the case, is that solvable via the xtypes dynamic APIs?

Karsten1987, 2019-10-22:
> So what I'm hearing is that a publisher on your ROS2 network may publish a type you do not have installed on the same computer as rosbag2, and in that case you will not be able to understand it.

That is correct

> That being the case, is that solvable via the xtypes dynamic APIs?

I believe that is the idea of xtypes. But at the same time I think it's neither settled in the standard of DDS nor really exposed through the ROS2 API. Unfortunately, I don't have enough insights in that to give you a really qualified answer.

emersonknapp, 2019-10-23:
Thanks for the clarification. Was that use case something that rosbag in ROS1 could handle? If not, I say that we close this issue and we can start to think about how important this feature is and how it might be implemented as a broader design discussion.

Karsten1987, 2019-10-23:
ros1 could handle this. The way I would try to explain this is that ROS1 could subscribe to any topic given that the _typesupport_ (aka TCPROS, it's not really a typesupport) was basically the same for every message. ROS1 can however only replay the messages it has locally stored as far as I remember. But again, I might be wrong here.

As for this ticket, I believe that the feature you're describing here is far broader than the scope of rosbag2. I'll close this for now, but we can re-iterate over it.

closed 2019-10-23 by Karsten1987