ros2/rosbag2 #197: Move get_storage_identifier and get_bagfile_size into BaseInfoInterface

zmichaels11, 2019-10-24:
As per https://github.com/ros2/rosbag2/pull/184#discussion_r338327221

As per the comment, these API calls are probably better if defined in `BaseInfoInterface` rather than `BaseIOInterface`.

### Required Changes
- [ ] Move `get_storage_identifier`
- [ ] Move `get_bagfile_size`
- [ ] Update `SqliteStorage`
- [ ] Update `ros2/rosbag2_bag_v2`

Karsten1987, 2019-11-18:
@zmichaels11 are you still working on this?

zmichaels11, 2019-11-18:
Yes I am, I'll push my branch once I rebase onto master.

zmichaels11, 2019-11-18:
Addressing `Move get_storage_identifier from BaseInfoInterface to BaseIOInterface` and `Move get_bagfile_size from BaseInfoInterface to BaseIOInterface` in PR: https://github.com/ros2/rosbag2/pull/209

zmichaels11, 2019-11-19:
Removed requirement `remove get_metadata` due to bug resolved by #210.

As per https://github.com/ros2/rosbag2/pull/210#issuecomment-555275503, we should probably discuss on handling `get_metadata` since its required by bagfiles not written by `Writer`.

zmichaels11, 2019-11-25:
Remove `get_metadata` was moved into https://github.com/ros2/rosbag2/issues/213 due to being more complex and requiring planning.

zmichaels11, 2019-12-02:
Corrected the tasks in OP;
'get_storage_identifier` and `get_bagfile_size` now correctly state that they're moving INTO `BaseInfoInterface`

closed 2019-12-06 by zmichaels11