ros2/rosbag2 #201: Recording of images fails

teutobod, 2019-11-08:
I'm publishing image data from my webcam using:
`ros2 run image_tools cam2image` [(link)](https://index.ros.org/doc/ros2/Tutorials/Quality-of-Service/)

The images are published to a topic of type:
`sensor_msgs/msg/Image`

Then I start recording using:
`ros2 bag record /image`
 _[INFO] [rosbag2_storage]: Opened database 'rosbag2_2019_11_08-16_48_23'.
[INFO] [rosbag2_transport]: Listening for topics...
[INFO] [rosbag2_transport]: Subscribed to topic '/image'
[INFO] [rosbag2_transport]: All requested topics are subscribed. Stopping discovery..._

But finally the recorded bag is empty!
_`ros2 bag info rosbag2_2019_11_08-16_48_23/`
Files:             rosbag2_2019_11_08-16_48_23.db3
Bag size:          16.3 KiB
Storage id:        sqlite3
Duration:          0.0s
Start:             Jan  1 1970 01:00:00.0 (0.0)
End                Jan  1 1970 01:00:00.0 (0.0)
Messages:          0
Topic information:_ 

**System: Ubuntu 18.04.3 LTS
ROS2 Release: Eloquent**

Karsten1987, 2019-11-11:
Could that be related to https://github.com/ros2/rosbag2/issues/125? Can you verify with which QoS you're sending the images? If the topics are sent reliably, I would assume the topics being correctly recorded. However, when you send images via unreliable QoS rosbag2 currently can't record this.

peterchen25, 2019-11-11:
According to [here](https://github.com/ros2/demos/blob/master/image_tools/src/cam2image.cpp#L169) the default qos of this demo in reliable

Karsten1987, 2019-11-12:
I think that wasn't actually accurate and [just got fixed](https://github.com/ros2/demos/pull/411).

With that fix - sending images reliably - I am able to record images:
```
 ➭ ros2 bag info rosbag2_2019_11_11-17_01_24/

Files:             rosbag2_2019_11_11-17_01_24.db3
Bag size:          59.3 MiB
Storage id:        sqlite3
Duration:          8.838s
Start:             Nov 11 2019 17:01:24.791 (1573520484.791)
End                Nov 11 2019 17:01:33.629 (1573520493.629)
Messages:          269
Topic information: Topic: /image | Type: sensor_msgs/msg/Image | Count: 269 | Serialization Format: cdr
```

teutobod, 2019-11-12:
Thanks for the update Karsten. Works now for me.

closed 2019-11-12 by teutobod