ros2/rosbag2 #202: LaserScan cannot be recorded

luyang00, 2019-11-11:
Hi,
I'm trying to record a rosbag for slam testcase. In my test, IMU message has been recorded and replay normally but sensor_msgs/laserscan seems not recorded properly. The replay of this bag only have IMU message. 
Then I try to only record laserscan message, the data size of this bug still not grow, so I guess it didn't write into a rosbag file but the log from rosbag2 on screen shows this topic has been subscribed successfully.  
Test on rosbag2 binary version at ros2-dashing.

Karsten1987, 2019-11-11:
I think this is a duplicate of https://github.com/ros2/rosbag2/issues/201.
I'll close this topic as I believe the matter can be discussed on the other issue. Feel free to re-open this ticket if you feel that it's fundamentally different.

closed 2019-11-11 by Karsten1987

mkhansen-intel, 2019-12-04:
I'm running into this same issue, the /scan topic is being published by Gazebo, but I can't record it.

The issue you linked looks different to me, in that the fix was to the demo being used. I think this actually duplicate of #125 

I'll comment there