ros2/rosbag2 #213: Resolve get_metadata in io-interfaces

zmichaels11, 2019-11-22:
Right now, `get_metadata` is defined in `BaseInfoInterface` and is only required in reading bagfiles that do not have a metadata file.

Issue #197 initially suggested removal of this function, however removal will cause breakage in `rosbag2_bag_v2`

I think this function should be removed since it exists only for compatibility and is required to be implemented by all plugins. Therefore some other mechanism should also be implemented to retain compatibility.

wjwwood, 2019-12-12:
@zmichaels11 I'll assign you for our triage process, feel free to reassign later.

zmichaels11, 2019-12-12:
> @zmichaels11 I'll assign you for our triage process, feel free to reassign later.

I'll create the required tickets to pull it into our sprint backlog.

Karsten1987, 2020-04-13:
For bookmarking, I think also `get_all_topics_and_types` should be deprecated in the storage interface.

Taken from https://github.com/ros2/rosbag2/pull/371#issuecomment-613146317
> The BaseReaderInterface provides get_all_topics_and_types, but this calls into the storage, which implements this as a database read

MichaelOrlov, 2025-05-07:
- Closing as wouldn't be done. We need `get_metadata()` and use it from multiple places, and we are not going to remove it.