ros2/rosbag2 #222: test_ros2_message segfaults when sanitizers disabled

jpsamper2009, 2019-12-02:
## Current Behavior

After compiling rosbag2 with `-DDISABLE_SANITIZERS=ON`, test_ros2_message segfaults:

```
5: Test command: /usr/bin/python3 "-u" "/opt/ApexOS/share/ament_cmake_test/cmake/run_test.py" "/builds/ApexAI/grand_central/apex_ws/build/rosbag2/test_results/rosbag2/test_ros2_message.gtest.xml" "--package-name" "rosbag2" "--output-file" "/builds/ApexAI/grand_central/apex_ws/build/rosbag2/ament_cmake_gmock/test_ros2_message.txt" "--command" "/builds/ApexAI/grand_central/apex_ws/build/rosbag2/test_ros2_message" "--gtest_output=xml:/builds/ApexAI/grand_central/apex_ws/build/rosbag2/test_results/rosbag2/test_ros2_message.gtest.xml"
5: Test timeout computed to be: 500
5: Running with ROS_DOMAIN_ID 1
5: -- run_test.py: invoking following command in '/builds/ApexAI/grand_central/apex_ws/src/apex_os/tools/rosbag2/rosbag2':
5:  - /builds/ApexAI/grand_central/apex_ws/build/rosbag2/test_ros2_message --gtest_output=xml:/builds/ApexAI/grand_central/apex_ws/build/rosbag2/test_results/rosbag2/test_ros2_message.gtest.xml
5: Running main() from gmock_main.cc
5: [==========] Running 11 tests from 1 test case.
5: [----------] Global test environment set-up.
5: [----------] 11 tests from Ros2MessageTest
5: [ RUN      ] Ros2MessageTest.allocate_ros2_message_allocates_basic_types_message
5: [       OK ] Ros2MessageTest.allocate_ros2_message_allocates_basic_types_message (1 ms)
5: [ RUN      ] Ros2MessageTest.allocate_ros2_message_allocates_strings_message_with_empty_string
5: [       OK ] Ros2MessageTest.allocate_ros2_message_allocates_strings_message_with_empty_string (0 ms)
5: [ RUN      ] Ros2MessageTest.allocate_ros2_message_allocates_strings_message_with_big_string_no_SSO
5: free(): invalid next size (normal)
5: -- run_test.py: return code 134
5: -- run_test.py: generate result file '/builds/ApexAI/grand_central/apex_ws/build/rosbag2/test_results/rosbag2/test_ros2_message.gtest.xml' with failed test
5: -- run_test.py: verify result file '/builds/ApexAI/grand_central/apex_ws/build/rosbag2/test_results/rosbag2/test_ros2_message.gtest.xml'
5/6 Test #5: test_ros2_message ................***Failed    0.27 sec
```

- ci.ros2.org doesn't do a "no sanitizer" build, which could explain why this wasn't caught earlier
- We were trying to disable the sanitizer on aarch64 because the sanitizer was unable to run in Gitlab CI using a docker container launched with Docker version 18.09 or older (e.g. https://gitlab.com/jpsamper/sanitize_aarch64/-/jobs/363319248)
    - Note that we were able to reproduce the segfault on x86_64 too, though

Karsten1987, 2019-12-02:
I am unable to reproduce this for dashing as I remember from an offline discussion that the tests were failing for you on dashing. Independent of the `DISABLE_SANITIZERS` flag, the tests run correctly on my Linux VM. 

To verify that the sanitizers work correctly, I've updated one of the tests by adding 
```
auto i = new int(42)
```
which - when running the tests - yield the expected outcome:
```
=================================================================
==18449==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 4 byte(s) in 1 object(s) allocated from:
    #0 0x7f5385a38c3b in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/liblsan.so.0+0xfc3b)
    #1 0x563a2c901900 in Ros2MessageTest_allocate_ros2_message_allocates_bounded_sequences_message_Test::TestBody() (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x12900)
    #2 0x563a2c97f35f in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x9035f)
    #3 0x563a2c9791e6 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x8a1e6)
    #4 0x563a2c9573f7 in testing::Test::Run() (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x683f7)
    #5 0x563a2c957d8c in testing::TestInfo::Run() (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x68d8c)
    #6 0x563a2c958435 in testing::TestCase::Run() (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x69435)
    #7 0x563a2c9634d3 in testing::internal::UnitTestImpl::RunAllTests() (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x744d3)
    #8 0x563a2c9804a6 in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x914a6)
    #9 0x563a2c979fe4 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x8afe4)
    #10 0x563a2c961f5f in testing::UnitTest::Run() (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x72f5f)
    #11 0x563a2c94f320 in RUN_ALL_TESTS() (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x60320)
    #12 0x563a2c94f2af in main (/home/karsten/ros2_dashing_ws/build/rosbag2/test_ros2_message+0x602af)
    #13 0x7f53846d5b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

SUMMARY: LeakSanitizer: 4 byte(s) leaked in 1 allocation(s).
```

Running the tests with valgrind led to a similar outcome. So I would assume that the Linux CI jobs pick up the sanitizers correctly as well on the buildfarm. Obviously, OSX and MSVC aren't effected by that since they don't use `gcc`. The latter makes me wonder why we haven't seen the segfaults happen on the buildfarm machines.

I am not really sure I understand your CI job correctly, but if you have a simple x86 docker image for me or an easy way of reproducing this, I am happy to help. 
Can you try to debug and provide a backtrace for where it is failing for you case? Maybe that can give some insights to pinpointing the problem.

michael-orlov1, 2019-12-04:
Hi Karsten,
You don't see failure on your CI buildfarm, because you are actually declaring and treating upper bounded strings and types as unbounded.
This particular failure relates to the issue in `/rosbag2/rosbag2/src/rosbag2/types/introspection_message.cpp` in function :
```
void allocate_element(
  void * data, const rosidl_typesupport_introspection_cpp::MessageMember & member)
{
  if (member.type_id_ == rosidl_typesupport_introspection_cpp::ROS_TYPE_STRING) {
    // This is necessary because initialization of empty strings fails for g++ compiled builds
```
There no special handling for constructing upper bounded strings. \
Failure happenning when  `if (member.string_upper_bound_)` and we are trying to construct upper bounded string as `std::string` in place with preallocated chunk of memory, which causing memory corruption. \
The same situation should be properly handled in cleanup_ellement(..)
Ar the moment we don't have generic solution for this problem which could be up-streamed.

michael-orlov1, 2019-12-04:
We also getting memory corruption in following test cases
```
TEST_F(Ros2MessageTest, allocate_ros2_message_allocates_bounded_sequences_message) {
  auto message = get_allocated_message("test_msgs/BoundedSequences");
```
and 
```
TEST_F(Ros2MessageTest, allocate_ros2_message_allocates_multi_nested_message) {
  auto message = get_allocated_message("test_msgs/MultiNested");
```
"Highly likely" for the same reason.

emersonknapp, 2020-01-29:
Closing for now as can't repro - can reopen if it continues to be an issue

closed 2020-01-29 by emersonknapp