ros2/rosbag2 #224: Allow only squashed PRs

piraka9011, 2019-12-02:
To ensure the quality of the commit history, we should only allow squashed commits from PRs. This is a configuration that can be enabled in the settings of the repo.

dirk-thomas, 2019-12-02:
In other repos (e.g. `rcpputils`) we have that option enabled. From my experience though I had to disable the flag temporarily a couple of times when I needed to do a non-squash merge. Therefore I am usually leaning towards teaching the reviewers than strictly enforcing it. (Just my 2ct)

piraka9011, 2019-12-02:
> I had to disable the flag temporarily a couple of times when I needed to do a non-squash merge. 

It sounds like this is the exception rather than the rule. It's better IMO to err on the safe side and that way we don't end up with weird commit histories.

In the event we need to disable the flag, the maintainer can approve the exception for the PR.

dirk-thomas, 2019-12-02:
> In the event we need to disable the flag, the maintainer can approve the exception for the PR.

Actually a maintainer wouldn't be able to do so. You need admin access to the repo to change the settings. Afaik there is no way to override the setting on a per-PR level.

Karsten1987, 2019-12-02:
That is correct, there is no flexibility to temporarily force a merge commit on a PR level. 
I agree with Dirk, that maintainers should be better advised to use squash merge over merge commits. 
However, as the last three PRs are already _wrongly_ merged, I've decided to disable the function for now. Also, as the ROS-Tooling WG is now an official maintainer of this repo, we might want to prevent that from accidentally happening. 

I propose we roll with it and see if it's becoming a burden to work with in the future.

piraka9011, 2019-12-03:
Closing for now. Will adjust the changelog next release and leave commit history as is for now. 

Will cut another issue if we see problems down the line.

closed 2019-12-03 by piraka9011