ros2/rosbag2 #229: relative_file_paths depends on call site

ieperen3039, 2019-12-10:
ros version: Eloquent
The `relative_file_paths` field in the yaml file specifies the location of the data file relative to the location of the terminal when `ros2 bag play <file>` is called.

The following should demonstrate the problem:
```
cd ~
ros2 bag record -a -o demo
^C (terminate the recording)
cd Documents
ros2 bag play ~/demo
```
One gets the following error:
```
Could not open 'demo/demo_0.db3' with 'sqlite3'. 
Error: Failed to read from bag: 
File 'demo/demo_0.db3' does not exist!
```

I would expect the relative path to be relative to the yaml file instead.

Might be related to issue [#49](https://github.com/ros2/rosbag2/issues/49)

zmichaels11, 2019-12-10:
@piraka9011 how difficult would it be to modify the Reader to support this?

piraka9011, 2019-12-10:
I don't think supporting this is as good as implementing the proposed design in #49. 
Also, it seems this is an issue directly related to the parsing of the tilde `~` and not a bug related to the Reader directly since passing an absolute path to the directory works with me.
@ieperen3039 can you confirm this? (i.e. try to run `ros2 bag play $HOME/demo`)

Karsten1987, 2019-12-10:
Just to clarify, #49 is not meant to be exclusive. Other plugins, such as the ros1 bagfile support does not have a metdata.yaml file in the first place. 
Also I see this problem here as unrelated to whether the input is specified as a directory or yaml file. The relative files within the metadata definitely have to be parsed with respect to the yaml file location and that is currently not working. That's simply a bug in the current reader implementation as far as I can tell.

piraka9011, 2019-12-10:
> Other plugins, such as the ros1 bagfile support does not have a metdata.yaml file in the first place.

Good call, something I should've considered.

> The relative files within the metadata definitely have to be parsed with respect to the yaml file location.

I might be able to take a look at this sometime later this week (after compression).

ieperen3039, 2019-12-11:
> @ieperen3039 can you confirm this? (i.e. try to run `ros2 bag play $HOME/demo`)

Verified that calling it using an absolute path gives the same result (that is also how i encountered it in the first place).
> That's simply a bug in the current reader implementation
'feature'

PR #345 (2020-04-03): resolve relative file paths
fixes #229 

It basically takes the relative file paths and resolves them relativ to the base folder name, essentially the name of the rosbag to be replayed.
If one of these relativ files is actually in an absolute form, it will ignore it.
closed 2020-04-03 by Karsten1987