ros2/rosbag2 #232: Improve Python API

jacobperron, 2019-12-12:
Currently, there is limited Python support for rosbag2; only a few functions are exposed via a [C-extension](https://github.com/ros2/rosbag2/blob/e2c96761c1886d14dbc470146dfd05a2528a8c35/rosbag2_transport/src/rosbag2_transport/rosbag2_transport_python.cpp) for recording, playing, and printing info for bags. Similar to the Python API in ROS 1, it would be useful to have facilities for looping through bags, requesting a subset of topics, exporting to CSV, etc.

## Prerequisites 

In order to deal with ROS messages from bags in Python, we'll need support for serializing/deserializing messages. This can be done by exposing the `rmw_serialize` and `rmw_deserialize` messages in Python. It makes sense to me to do this in rclpy.

## Implementation considerations

We can extend the existing C-extension to expose more of the rosbag API and move it to it's own package, `rosbag2_py`. Then a new rosbag2 Python library can be built around it, using rclpy as needed.

For a first iteration, I think having the ability to read/write messages to/from bags from Python would be great; similar to [how this was done in ROS 1](http://wiki.ros.org/rosbag/Code%20API#py_api).

jacobperron, 2020-01-10:
I've implemented a proof-of-concept using pybind11 to read messages from a rosbag in a Python for loop.

Here's an [example](https://gist.githubusercontent.com/jacobperron/d2f34e74e43985c7d02ea925f23cafed/raw/da1a6fda909b66d1acd5049cf96ad38b4660d944/read_bag.py) of parsing a bag, which requires the following two branches of `rclpy` and `rosbag2`:

https://github.com/ros2/rclpy/tree/jacob/rclpy_serialize

https://github.com/ros2/rosbag2/tree/jacob/rosbag2_py

As well as `pybind11_vendor`:

https://github.com/ros2/pybind11_vendor

---

I think pybind11 looks like a promising tool for us to wrap the cpp implementation. Next, we should identify the data structures to expose/wrap in `rosbag2_py` and add a proper Python interface on top.

I'll open a PR to `rclpy` soon with the message serialization functions.

@Karsten1987 FYI

jacobperron, 2020-01-10:
`rclpy` PR adding serialization functions: https://github.com/ros2/rclpy/pull/495

Karsten1987, 2020-03-02:
@jacobperron Are you planning on working on a PR for the python API (aka `rosbag2_py`)?

mabelzhang, 2020-03-02:
I have a branch https://github.com/mabelzhang/rosbag2/tree/mabel/rosbag2_py built on top of Jacob's `jacob/rosbag2_py` branch here. It only does basic reading right now - example usage of all the functions is [here](https://gist.github.com/mabelzhang/31468244b50e6ebc5a7e374af0e01fec). Does it need to be more complete before a PR is opened?

Karsten1987, 2020-03-02:
I am asking because of https://github.com/ros2/rosbag2/pull/306#pullrequestreview-367431071
Currently the only entry point for `ros2bag` is `rosbag2_transport`. However, I feel that this should not be a requirement for some functionality such as `convert`. This could be handled solely within the `rosbag2_cpp` API and doesn't require any transport. For this rationale, we'd ideally have a `rosbag2_py` which provides such an entry point.

I don't expect `rosbag2_py` to be feature complete. But I'd prefer to have something going where we can extend functionality before trying to work around the lack of a Python API. Do you mind opening a PR with your current works?

mabelzhang, 2020-03-03:
Okay #308

emersonknapp, 2020-06-05:
Will that PR close this ticket? Or, are there more functionality that needs to be implemented to call this done? If so, could we make a checklist on the issue description?

mabelzhang, 2020-06-05:
We haven't merged that one because we want a writer along with the reader to make it more complete.
#419 provides a writer function and I have been reviewing it, but it would be good if one of the maintainers can take a look and move it along. It adds to #308.
Then we can merge both together when they're ready.

emersonknapp, 2020-06-05:
Yeah, I understand that part - but my question is more along the lines of - does the Reader+Writer python combo suffice to close this first pass of the "Improve Python API" ticket? Just trying to get a handle on what the completion criteria is for this

mabelzhang, 2020-06-05:
Ah. Oh my end I think so. @Karsten1987 might have more to add.

jacobperron, 2020-08-29:
I think we can close this, but feel free re-open or open a new ticket for additional features to the Python API.

Karsten1987, 2020-08-29:
I am not sure we should close this ticket as long as the `AMENT_IGNORE` file is present in the `rosbag2_py` package.
@mabelzhang @jacobperron do you have an estimate on when the python API can be used?

jacobperron, 2020-09-09:
@Karsten1987 AFAIK, it can be used. We've disabled the testing of it with the `AMENT_IGNORE` file until the following test failures can be resolved:

- [ ] https://github.com/ros2/rosbag2/issues/505
- [ ] https://github.com/ros2/rosbag2/issues/504

I'll reopen this until they are resolved and the `AMENT_IGNORE` file removed.

emersonknapp, 2020-10-26:
It looks like both linked issues have been resolved - I'll close this issue. Please reopen if needed

closed 2020-10-26 by emersonknapp