ros2/rosbag2 #233: Split bags by time when recording

emersonknapp, 2019-12-12:
Rosbag 1 had the ability to split bags while recording into chunks of approximately the same time duration.

The feature now exists to split by bag size while recording, but not by time.

This issue requests that the by-time splitting feature be added.

dsharma2701, 2020-04-28:
Just wondering if this is now available ?

emersonknapp, 2020-04-29:
No, this feature hasn't been started yet. We may land a few already in-flight features before the Foxy release (this package is not technically part of the core, so is not technically frozen yet), but are not planning to start any new feature work for this release so it ill probably be planned for the next ROS 2 distro.

Karsten1987, 2020-04-29:
There is infrastructure in place (namely in `rosbag2_cpp`) to split bag files by size. I think the step to enhance the functionality to split by time is pretty straight forward. It's essentially adding a new field [here](https://github.com/ros2/rosbag2/blob/master/rosbag2_cpp/include/rosbag2_cpp/storage_options.hpp#L29-L31) to give a timestamp, in seconds or nanoseconds, and adapt the logic in [here](https://github.com/ros2/rosbag2/blob/master/rosbag2_cpp/include/rosbag2_cpp/writers/sequential_writer.hpp#L126-L130). 
That is all to say, PRs are welcome ;-)

jhdcs, 2020-05-13:
I've started work on this in #409, but I'm having some issues getting the command line to recognize the new option. Is there any documentation about adding new commands to the command line? Or something obvious that I'm missing?

Kaju-Bubanja, 2021-01-11:
This issue can be closed now, since #409 is merged

emersonknapp, 2021-01-12:
> This issue can be closed now, since #409 is merged

Thanks for the ping!

closed 2021-01-12 by emersonknapp