ros2/rosbag2 #240: record_end_to_end_test unstable

zmichaels11, 2019-12-19:
`record_end_to_end_test` has a small chance of failing: 

```
  [1.194s] 1: [ERROR] [1576711769.035610056] [rosbag2_storage]: Could not open 'tmp_test_dir.jpS0UF/bag/bag_0' with 'sqlite3'. Error: Failed to setup storage. Error: Error processing SQLite statement. Return code: 5
  [1.195s] 1: [ERROR] [1576711769.035829055] [rosbag2_storage]: Could not load/open plugin with storage id 'sqlite3'.
  [1.195s] 1: [ERROR] [1576711769.035939955] [rosbag2_transport]: Failed to record: No storage could be initialized. Abort
```
This test has failed 3x total with a frequency of about 1/40 tests.

### Resources:
* [SQLite return code 5](https://www.sqlite.org/rescode.html#busy)
* ["Error processing SQLite statement"](https://github.com/ros2/rosbag2/blob/074055632c0d649050d3e25ab8d17b2c238cded8/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_statement_wrapper.cpp#L55)
* ["Failed to setup storage"](https://github.com/ros2/rosbag2/blob/074055632c0d649050d3e25ab8d17b2c238cded8/rosbag2_storage_default_plugins/src/rosbag2_storage_default_plugins/sqlite/sqlite_storage.cpp#L87)

alrashwa, 2020-01-02:
I'm seeing a similar error while following the ROS2 "Recording and Playing back data" tutorial. Any updates on this?

> ~/bag_files$ ros2 bag record /turtle1/cmd_vel
[ERROR] [rosbag2_storage]: Could not load/open plugin with storage id 'sqlite3'.
[ERROR] [rosbag2_transport]: Failed to record: No storage could be initialized. Abort

zmichaels11, 2020-01-02:
@alrashwa that error looks like its caused by not having the `rosbag2 sqlite3 storage plugin` installed. Its provided by the package `rosbag2_storage_default_plugins`. 

If that's not the cause, can you please respond back with more information (ROS2 distro, OS, type of ROS2 install) so I can determine if its related to this issue?

Thanks

alrashwa, 2020-01-03:
Thanks, I think that package was it. It looks like the rosbag2 install process as shown in the [tutorials ](https://index.ros.org/doc/ros2/Tutorials/Ros2bag/Recording-And-Playing-Back-Data/) usually would install this package. I ran into an issue (looks like others have as well) where the installer is interrupted when it hits a dependency on `ros-eloquent-ros1-bridge`. 

I was able to install the `rosbag2_storage_default_plugins` individually using `apt-get install ros-eloquent-rosbag2-storage-default-plugins ros-eloquent-ros2bag` and continue the tutorial.

zmichaels11, 2020-01-23:
I want to go through the build & test history and get a better metric at how frequent this fails.

zmichaels11, 2020-01-24:
Failures:
* https://github.com/ros2/rosbag2/runs/382603849?check_suite_focus=true (15d ago)
* https://github.com/ros2/rosbag2/runs/362818450?check_suite_focus=true (Dec 24)
* https://github.com/ros2/rosbag2/runs/355436886?check_suite_focus=true (Dec 18)

zmichaels11, 2020-01-31:
https://github.com/ros2/rosbag2/pull/278

hazkaz, 2020-06-08:
For complete noobs who might have made the mistake I did - the tutorial asked me to install ros2bag using this command

`sudo apt-get install ros-<distro>-ros2bag ros-<distro>-rosbag2*`

I ended up typing the last part as **ros2bag*** instead of **rosbag2***

emersonknapp, 2021-05-14:
It seems that mostly this was an issue in the documentation and/or user error - that should be resolved now, as you can just `apt-get install ros-<distro>-rosbag2` and get all the required packages.

If this is a flaky `recording` test - we'll roll that into the work for https://github.com/ros2/rosbag2/issues/732

closed 2021-05-14 by emersonknapp

revanthsenthil, 2023-02-12:
@zmichaels11 I have the exact same issue that you initially commented, but none of these fixes fixed it. I am just trying to run `ros2 bag record <topic>`. Any fixes on this?

ronyshaji, 2023-06-30:
> @zmichaels11 I have the exact same issue that you initially commented, but none of these fixes fixed it. I am just trying to run `ros2 bag record <topic>`. Any fixes on this?

Did you able to find a solution for this ?