ros2/rosbag2 #242: Remove requirement for ros1-bridge requirement for ros2 bag command

kellyprankin, 2019-12-30:
Is there any timeline on removal of the need for having ros1 installed to use the ros2 bag command?

Karsten1987, 2019-12-30:
Can you elaborate a bit on where you see this dependency? The only package which depends on the ros1 bridge is the [rosbag2_bag_v2_plugin](https://github.com/ros2/rosbag2_bag_v2).

The current packages within this repo are independent of the bridge.

kellyprankin, 2019-12-30:
Maybe I am incorrect. When I try to install the ros2 bag command it says that it depends on the ros1-bridge. Did I install the wrong one?

Karsten1987, 2019-12-30:
It depends how you installed your packages. All packages you find on the eloquent branch of this repo can be installed via apt on your Linux system. 
If you have installed all packages based on a ‘rosbag2’ prefix you most likely also installed the before mentioned plugin for ros1 bag files.

kellyprankin, 2019-12-30:
@Karsten1987 The instructions at https://github.com/ros2/rosbag2 say to do:

 sudo apt-get install ros-eloquent-ros2bag* ros-eloquent-rosbag2*

Karsten1987, 2019-12-30:
This way will install all packages including the ros1 bridge related one. 
In order to avoid that you have to install the individual packages by hand for now. That is:
ros-eloquent-rosbag2-transport and ros-eloquent-ros2bag 

We are currently working making this easier and the ros1 bridge related one really only optional: #241

kellyprankin, 2019-12-30:
Thanks for talking me through it.  I'll give that a shot

kellyprankin, 2019-12-31:
Your solution worked for me. I am hoping I can get them to update the documentation for the ROS2 Tutorial, because it isn't clear that you can avoid the issue. Thanks again

dirk-thomas, 2020-01-02:
> I am hoping I can get them to update the documentation for the ROS2 Tutorial, because it isn't clear that you can avoid the issue.

The instructions are in the README file in this repo: https://github.com/ros2/rosbag2/blob/master/README.md

@kellyprankin Please consider to propose a pull request to improve the instructions.

closed 2020-01-06 by Karsten1987