ros2/rosbag2 #266: error: ‘rcutils_calculate_directory_size’ was not declared in this scope [Unable to build rosbag2 from source]

cardboardcode, 2020-01-26:
Hi, thank you for providing implementation for one of most useful tools in ROS. 

# Issue Description
I attempted building **rosbag2** from source and encountered the following error upon running `colcon build`.

> `error: ‘rcutils_calculate_directory_size’ was not declared in this scope`

I am thus unable to build and use **rosbag2**. 

# Context
My **objective** is to run the command `$ ros2 bag info <bag.file>` without the following error as highlighted in #94 :

> `[Failed to load entry point 'info': /opt/ros/dashing/lib/librosbag2_storage.so: undefined symbol: _ZNK12class_loader11ClassLoader15isLibraryLoadedEv
The C extension '/opt/ros/dashing/lib/python3.6/site-packages/rosbag2_transport/_rosbag2_transport_py.cpython-36m-x86_64-linux-gnu.so' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Segmentation fault (core dumped)](https://github.com/ros2/rosbag2/issues/94)`

This error is encountered when **installation is done via the debian packages**, based on the instructions given in this repository's [readme](https://github.com/ros2/rosbag2#debian-packages).

# Environment
- ROS Dashing
- Ubuntu 18.04.3

Any help is greatly appreciated. If more information is needed, feel free to ask.

zmichaels11, 2020-01-27:
Hi,
Which branch are you building from?

cardboardcode, 2020-01-28:
Hi @zmichaels11, 

I built from `master` branch.

zmichaels11, 2020-01-28:
Have you tried building the `dashing` branch? I believe `master` should only be used if you're building all of ROS2 from source.

cardboardcode, 2020-01-28:
Will give a try today and update again on this thread as soon as possible.

Thank you for the suggestion, @zmichaels11 :)

cardboardcode, 2020-01-28:
# [Solution]
Thanks @zmichaels11. **The issue is resolved.** 

I have tried building rosbag2 from source using the branch `dashing`. **The build error in this thread no longer appears.** The build successfully finishes.

Follow the instructions below to reproduce the successful build of **rosbag2**.

> mkdir -p ~/ros2_ws/src && cd ~/ros2_ws/src

> git clone --single-branch --branch dashing https://github.com/ros2/rosbag2

> cd ~/ros2_ws

> source /opt/ros/dashing/setup.bash 

> colcon build

@zmichaels11 May I make a quick **Pull Request** to update the documentation on **master** branch? With clearer instructions, hopefully less would run into this issue.

closed 2020-01-28 by cardboardcode

zmichaels11, 2020-01-28:
Yes, please!

And thanks for reporting back with the steps required to use `rosbag2` from source on `Dashing`.

commit 430c4e5d (2020-01-31): Fixed #266 with updated documentation (#272)