ros2/rosbag2 #271: Resolve workarounds in rosbag2_tests for Windows

zmichaels11, 2020-01-28:
`test_rosbag2_record_end_to_end` defines a workaround for Windows in multiple places due to Windows not supporting a mechanism to programmatically send a SIGINT.

These tests should either be rewritten to not require the workaround or a helper function should be written so less platform-specific code is required.

piraka9011, 2020-02-07:
I have created a Pytest that can successfully execute a `ros2bag` CLI command and send a `CTRL_C_EVENT` to gracefully terminate it. The issue is that when I run it with `colcon test` the `CTRL_C_EVENT` aborts the `colcon test` command as well.

I opened an issue (colcon/colcon-core#303) to discuss why this is the case and propose a path forward. 

An potential approach in the future would be to use `stdin` pipes to communicate with the `ros2bag` command via keyboard events if that feature is developed.

zmichaels11, 2020-02-14:
As per our discussion yesterday, it might make sense to rewrite the e2e tests in Python and use launch.

Karsten1987, 2020-07-01:
I think one helpful addition would be to introduce a `close` function to the final `Reader` and `Writer` class. The closing function is generally provoked via the destructor, which works well as long as you don't try to `ctrl-c` on Windows. Having a public function for this would allow us to manually trigger this within the tests before `ctrl-c`.

MichaelOrlov, 2024-07-13:
- Closing as a stale issue.
- The helper functions are already inplace, and tests will be rewritten to be able to send analog of the CTL+C on Windows in the scope of the https://github.com/ros2/rosbag2/pull/1342

closed 2024-07-13 by MichaelOrlov