ros2/rosbag2 #274: Compress files & messages aynchronously or in a seperate thread

piraka9011, 2020-01-30:
Original request  from #152:
> Compression is applied to each message asynchronously in parallel by a thread from a thread pool (with command line option to specify number of compression threads).

-----

Compression currently occurs in a single threaded workflow. To avoid potentially dropping messages from the message queue, compression should occur asynchronously or in a separate thread.

pjreed, 2020-08-11:
I'd be interested in working on this if nobody else is doing so right now.  Thanks!

Karsten1987, 2020-08-11:
sounds good to me!

emersonknapp, 2020-08-12:
@pjreed nice! I've assigned the ticket you and moved the issue to In Progress on the ZenHub board.

pjreed, 2020-08-14:
I've got a working implementation for this, although while reviewing it, I realized there are a couple other things we might want to address.

If the incoming queue overflows, the result is that in message compression mode, the messages will be lost, and in file compression mode, the file will still exist but will not be compressed.  That's not really worse than the current behavior, which will end up dropping messages while the main thread is compressing, but I think it would be preferable for messages & files to be stored uncompressed rather than completely dropped.

The catch here is that the current format seems to assume that if compression is enabled, then *all* messages or files are compressed.  I haven't actually reproduced the condition in order to test it, but I suspect something would crash or otherwise throw an exception when trying to read bags with mixed-compression data.  What do you think a reasonable approach to handling this would be?  We could leave that as-is for now and open another issue to address it, or also have this PR handle playing mixed-compression bags, or if somebody else has another idea...

piraka9011, 2020-08-14:
> I suspect something would crash or otherwise throw an exception when trying to read bags with mixed-compression data.

Correct, most likely what will happen is that when you compress/decompress an uncompressed message/file, Zstd will throw an error since it's not in the format it expects. (Zstd adds some frame metadata to compressed objects).

I don't know if it's within the scope of rosbag2 to handle mixed compression data (maybe it can be a CLI option...?) but one way could be checking the error type (list [here](https://github.com/facebook/zstd/blob/0ed07f6dfe942627e7724e5b910e23ec3cd8ece8/lib/common/zstd_errors.h#L39)) and if it's a frame/checksum issue then we can try to read the message/file uncompressed.

emersonknapp, 2020-08-19:
From the conversation in https://github.com/ros2/rosbag2/pull/511 - we're thinking about trying to consolidate this compression writer logic with the regular writer logic. If we took that approach, we could have a single queue-processing mechanism on threads (rather than making it compression-specific)

The thread receiving messages would simply toss them on the queue, then the worker thread would take  care of convert->compress->write. This structure would also help with https://github.com/ros2/rosbag2/issues/436

Then I think there would be no possibility of "mixed-compression". Either the queues can keep up (and the messages _must_ be converted+compressed before being written to storage), or messages start being dropped entirely.

pjreed, 2020-09-04:
Since #511 looks like it's still under development, do you think it'd be a good idea to go ahead and merge this in for now and then replace it later, or would it be better just to wait for that to be finished?

emersonknapp, 2020-09-04:
We should go ahead with your change - the time frame on that restructuring is uncertain, we're pulled in a few directions at the moment. Sorry to make you wait on that, I had high hopes heh

PR #506 (2020-12-04): Compress bag files in separate threads
This offloads all compression into separate threads.  Doing so helps
to prevent rosbag2 from dropping messages due to the main thread
being busy, and it also helps to improve performance by spreading
the work across multiple CPU cores.  It uses a producer/consumer
model with a fixed number of pre-allocated threads that consume
messages or files off of an incoming queue.

This adds two new command line options:
- --compression-queue-size
   - The number of messages or files that can be sitting in the queue
     waiting for a thread to consume them.  If the [... cut]
commit 2d45abb1 (2020-12-04): Compress bag files in separate threads (#506)

* Compress bag files in separate threads

This offloads all compression into separate threads.  Doing so helps
to prevent rosbag2 from dropping messages due to the main thread
being busy, and it also helps to improve performance by spreading
the work across multiple CPU cores.  It uses a producer/consumer
model with a fixed number of pre-allocated threads that consume
messages or files off of an incoming queue.

This adds two new command line options:
- --compression-queue-size
   - The number of messages or files that can be sitting in the queue
 [... cut]
closed 2020-12-04 by emersonknapp