ros2/rosbag2 #275: Compression options for high throughput environments

piraka9011, 2020-01-30:
Original request in #152:
> Options for what to do if messages are coming in faster than they can be compressed - drop messages or store uncompressed versions.

-----

A potential approach is adding a CLI flag that specifies how to handle the messages, where the default behaviour is to store uncompressed versions.

Example: Passing the `--drop-msg` flag means that rosbag2 is allowed to drop messages

```sh
ros2 bag record -a --compression-mode message --drop-msg
```

If it is not passed, then rosbag2 will simply store an uncompressed version of the message.

For better long term support in case we need to add more compression options, something along the lines of:

```sh
ros2 bag record -a --compression-mode message --compression-options [key=value, key=value]
```

might be a better approach.

MichaelOrlov, 2021-05-06:
@piraka9011 If we would add option for storing uncompressed message if messages coming faster than they can be compressed, how we will distinguish on playback if message wasn't compressed to avoid applying decompression on it?

piraka9011, 2021-05-07:
Most compression libs I've seen have some way of identifying if some chunk of data has been compressed or not.

For ex. [`zstd`](https://facebook.github.io/zstd/zstd_manual.html#Chapter3) throws a `ZSTD_CONTENTSIZE_ERROR` when attempting to decompress if the data doesn't have a valid initial frame (which in the docs states is a magic number).

Here's an example: https://github.com/facebook/zstd/blob/dev/examples/simple_decompression.c

This assumption probably doesn't hold to all compression libs but is probably one way to start looking at addressing your concern.

MichaelOrlov, 2021-05-07:
Thanks for clarification. Might be a valid workaround.

emersonknapp, 2022-08-20:
Closing this issue as nonspecific. It's more of a discussion rather than a specific enhancement request. Since it was first opened there has been a lot of work on the compression infrastructure, so the discussion is a bit out of date.

If there is a specific feature we should implement, we can open a new issue or reopen this one with concrete completion criteria.

closed 2022-08-20 by emersonknapp