ros2/rosbag2 #288: Cannot specify log level via the CLI

piraka9011, 2020-02-07:
There is currently no mechanism to specify the log level ros2bag via the CLI.

I have attempted to configure the level using `__log_level:=debug` and `--log-level` flags, however they both are assumed to be topics by Python's `argparser`.

This may require specifying the log level in `ros2bag` and propogating it accordingly to `rosbag2_transport` where we could then use `rcutils_logging_set_logger_level()` with the `transport_node`.

emersonknapp, 2021-05-14:
Now that the Recorder and Player are Nodes, we should be pretty close to being able to configure the log level via parameters to those nodes. Though, the workflow is not quite connected yet.

MichaelOrlov, 2024-12-22:
- Closing as resolved by https://github.com/ros2/rosbag2/pull/1625

closed 2024-12-22 by MichaelOrlov