ros2/rosbag2 #290: Playback rate / speed option

emersonknapp, 2020-02-07:
## Description

`rosbag` had the option `-r` to control the playback speed. This would be a great feature in rosbag2!

## Related Issues

N/A

## Completion Criteria

* A user can specify a commandline option to `ros2 bag play` to control the speed of the playback
* The speed may be a float > 0
* The playback does not strictly have to meet the requested speed if it can't play back that fast, but it would be nice if it gave some indication of the actual speed of playback (this may be a separate feature however, and possibly useful in `-r 1` scenarios as well)

## Implementation Notes / Suggestions

Multiply the incrementing time counter used to query messages from the bag by a constant factor, instead of directly using the wall time.

## Testing Notes / Suggestions

Playback a bag with the rate option enabled, with a single subscriber to a topic that counts the number of messages received, then assert that they are within an expected range
* Test with rate = 2
* Test with rate = 0.5

Karsten1987, 2020-03-31:
@emersonknapp @mabelzhang can this ticket be closed with #304 being merged?

emersonknapp, 2020-04-01:
Yep!

closed 2020-04-01 by emersonknapp