ros2/rosbag2 #296: Cannot play multiple compressed files

dabonnie, 2020-02-14:
## Description

When attempting to play back rosbag2 compressed data, which multiple files, the playback terminates with an error.

### Expected Behavior

Multiple compressed bag files should be able to be played back.

### Actual Behavior

After recording rosbag data with the command

```
ros2 bag record -a -b 134217728 --compression-mode file --compression-format zstd
```

and attempting to play back the resulting files (more than 1) with the command

```
ros2 bag play <my_bag_directory>
```

it appears the first bag file is decompressed and play, but when attempting to decompress the ensuing file the following error is printed to the console:

```
ERROR] [1581565911.633620497] [rosbag2_transport]: Failed to play: Bagfile is not opened 
```

## To Reproduce

1. Record compressed data which results in multiple bag files
2. Attempt to play the recorded data
3. Console  / log print error results in error ```[rosbag2_transport]: Failed to play: Bagfile is not opened ```

## System (please complete the following information)
 - OS: ubuntu 18.04
 - ROS 2 Distro: master, building from source

dabonnie, 2020-02-14:
This is a potential fix: https://github.com/ros2/rosbag2/pull/294

emersonknapp, 2020-02-14:
Thanks for reporting this as an issue - makes it easier to track

closed 2020-03-12 by zmichaels11