ros2/rosbag2 #298: Undefined symbols for architecture for YAML::Node::Scalar() const in metadata_io.cpp.o

tanzby, 2020-02-15:
## Description

followed the steps of installing ros2 here: [https://index.ros.org/doc/ros2/Installation/Dashing/OSX-Install-Binary/](https://index.ros.org/doc/ros2/Installation/Dashing/OSX-Install-Binary)  and try to build `rosbag2_storage`

```
colcon build --symlink-install --packages-select rosbag2_storage
```
Note that `yaml_cpp_vendor` has already been setup up.

[image]

### Expected Behavior

Build Package successfully.

### Actual Behavior

```
➜  colcon build --symlink-install --packages-select rosbag2_storage
Starting >>> rosbag2_storage
--- stderr: rosbag2_storage                              
Undefined symbols for architecture x86_64:
  "YAML::detail::node_data::empty_scalar()", referenced from:
      YAML::Node::Scalar() const in metadata_io.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [librosbag2_storage.dylib] Error 1
make[1]: *** [CMakeFiles/rosbag2_storage.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rosbag2_storage	[ Exited with code 2 ]

Summary: 0 packages finished [12.0s]
  1 package failed: rosbag2_storage
  1 package had stderr output: rosbag2_storage
```

## System
 - OS: macOS 10.15.3
 - ROS 2 Distro: Dashing
 - Version: master branch

tanzby, 2020-02-16:
I upgrade the` yaml-cpp` version in `yaml_cpp_vendor` to `0.6.3` by modifying CMakeLists.txt to the following: 

```
ExternalProject_Add(yaml_cpp-9a36242
    URL https://github.com/jbeder/yaml-cpp/archive/yaml-cpp-0.6.3.zip
    URL_MD5 e8a182537af663cc45228f7064b2021c
    TIMEOUT 600
    LOG_CONFIGURE ${should_log}
    LOG_BUILD ${should_log}
    CMAKE_ARGS
      -DCMAKE_INSTALL_PREFIX=${CMAKE_CURRENT_BINARY_DIR}/yaml_cpp_install
      ${extra_cmake_args}
      -Wno-dev
  )
```
Then I re-build `yaml_cpp_vendor` then `rosbag2_storage`,  it works.  Note that the original version of `yaml_cpp_vendor` has a yaml-cpp with version `0.6.1` and my macOS was installed with a 0.6.3 yaml-cpp by Homebrew. If the consistency of System's and yaml_cpp_vendor's dose the impact? e.g using a system's `<yaml-cpp/yaml-cpp.hpp>` but link with yaml_cpp_vendor's dylib.

Karsten1987, 2020-04-01:
@tanzby this rather looks like an issue better suited for https://github.com/ros2/yaml_cpp_vendor, could that be? I am how `rosbag2_storage` could behave differently.

Karsten1987, 2020-05-06:
I am closing this issue due to inactivity. @tanzby please feel free to re-open in case your problem still persists.

closed 2020-05-06 by Karsten1987