ros2/rosbag2 #305: Bag splitting test is unreliable with zstd_file_compression on Windows

brawner, 2020-03-01:
The `record_end_to_end_test_with_zstd_file_compression_compresses_files` test has regular failures on `nightly-win-rep` jenkins builds. It will periodically fail to split the bag file. This test is fairly easy to reproduce with a build like [this](https://ci.ros2.org/job/ci_windows/9419/). 

Example:
https://ci.ros2.org/view/nightly/job/nightly_win_rep/1770/testReport/rosbag2_tests/RecordFixture/record_end_to_end_test_with_zstd_file_compression_compresses_files/

I have only seen this flaky failure occur on Windows platforms.

brawner, 2020-03-01:
Possibly related issues:
https://github.com/ros2/rosbag2/issues/271
https://github.com/ros2/rosbag2/issues/240

zmichaels11, 2020-03-11:
Related PRs:
* https://github.com/ros2/rosbag2/pull/315
* https://github.com/ros2/rosbag2/pull/294

Karsten1987, 2020-03-30:
@zmichaels11 @emersonknapp Is anybody from your group working on this? It's apparently a recurring issue on every windows build.

emersonknapp, 2020-03-30:
@dabonnie maybe, anything immediately obvious about this?

Karsten1987, 2020-05-08:
I just looked into one of the failing tests: `record_end_to_end_with_splitting_bagsize_split_is_at_least_specified_size`.

It fails with the assert in here:
https://github.com/ros2/rosbag2/blob/c28c89267fa8e22e76c6540ffb700f908eca04db/rosbag2_tests/test/rosbag2_tests/test_rosbag2_record_end_to_end.cpp#L281

And while we are filling the metadata by hand, the only reason this tests fails is that the individual paths are actually not available:
https://github.com/ros2/rosbag2/blob/master/rosbag2_tests/test/rosbag2_tests/test_rosbag2_record_end_to_end.cpp#L260-L268

That makes me believe it's not one of the `Ctrl-C` limitations but either a flaw in that particular windows logic or that the paths are indeed not existing which would make me believe that the splitting functionality does not work as expected on Windows.

Any ideas?

PR #407 (2020-05-12): Fix splitting tests on windows
fixes #305
closed 2020-05-12 by Karsten1987