ros2/rosbag2 #307: Ability to playback bags with unknown message types

clalancette, 2020-03-02:
## Description

Sometimes I am handed a bagfile that has messages I don't have the sources for.  Right now if I try to play those bag files back with `ros2 bag play`, I get an error message like:

```
[ERROR] [1583183596.808870285] [rosbag2_transport]: Failed to play: package 'custom_pkg' not found, searching: [/home/ubuntu/ros2_ws/install/rosbag2, /home/ubuntu/ros2_ws/install/ros2bag, ...]
```

It would be nice if I could still play back these bag files, and `ros2 bag play` would just play the ones it could deserialize.  Basically I am asking for this error to be downgraded to a warning, and skipping the messages that can't be played back.

I'm currently working around this by manually using sqlite3 to remove the messages I don't have the sources for.

## Related Issues

None.

## Completion Criteria

* [ ] Bag file with unknown messages can be played back

## Implementation Notes / Suggestions

Make the initial error messages about unknown messages a warning, and then skip those messages that can be deserialized during playback.

## Testing Notes / Suggestions

Make sure there is a bag file with unknown message types, and that it can be played back.

emersonknapp, 2020-03-02:
Good idea, perhaps an explicit argument to `play`, something like `--ignore-unknown-types`

clalancette, 2020-03-02:
> Good idea, perhaps an explicit argument to `play`, something like `--ignore-unknown-types`

I'd lean towards making it the default, with a warning, but a command-line option would work too.

kielczykowski, 2021-04-14:
Shouldn't this issue be closed due to https://github.com/ros2/rosbag2/pull/592 and https://github.com/ros2/rosbag2/pull/394 or is the implementation still needed?

emersonknapp, 2021-04-14:
Good catch - this is closed by #592

closed 2021-04-14 by emersonknapp