ros2/rosbag2 #309: How to play a ros1 bag file with a none CDR rmw implementation

hannatao, 2020-03-03:
I have a rmw implementation which serialization format is not CDR. When I try to record and play a ros2 application this works fine. Rosbag2 will use the current rmw implementation's serialization.

But when I try to play a ros1 bag file, it works fine with default fastRTPS. But when I try to use the specific rmw implementation, rosbag2 will show the following erros.
```
ros2 bag play -s rosbag_v2 2020-02-28-12-53-21.bag
[ERROR] [rosbag2]: Requested converter for format 'sCDR' does not exist
[ERROR] [rosbag2_transport]: Failed to play: Could not find converter for format sCDR
```
Is there any ideas of playing a ros1 bag file with a none CDR rmw implementation? Or I have to write a ros2 plugin with the current rmw implementation? 

Thanks.

Karsten1987, 2020-03-16:
The rosbag2_bag_v2_plugin has essential two components. A storage interface and a converter interface. The storage plugin really only opens a ROS1 bag file and extracts its serialized content. There is no conversion happening yet.
The converter is really just leveraging the ros1_bridge which helps to convert ROS1 message types to ROS2 messages. 

In which format do you want to replay your bag file? Does your custom RMW implementation support sending serialized messages?

hannatao, 2020-03-19:
I understand when I try to play a ROS1 bag file, the rosbag2 process includes reading bag file, deserializing ROS1 msg, converting ROS1 msg to ROS2 msg(all these are enabled by rosbag2_bag_v2_plugin), serializing ROS2 msg(This is what I want to ask), sending serialized msg.
The custom RMW implementation support rmw_publish_seialized_message and rmw_serialize interfaces. When trying to play ROS2 bag file, It works fine. But when trying to play ROS1 bag file, rosbag2 will ask to provide a converter for serializing ROS2 msg rather than using the custom rmw_serialize interface to serialize ROS2 msg.

hannatao, 2020-03-19:
> The rosbag2_bag_v2_plugin has essential two components. A storage interface and a converter interface. The storage plugin really only opens a ROS1 bag file and extracts its serialized content. There [...]

I understand when I try to play a ROS1 bag file, the rosbag2 process includes reading bag file, deserializing ROS1 msg, converting ROS1 msg to ROS2 msg(all these are enabled by rosbag2_bag_v2_plugin), serializing ROS2 msg(This is what I want to ask), sending serialized msg.
The custom RMW implementation support rmw_publish_seialized_message and rmw_serialize interfaces. When trying to play ROS2 bag file, It works fine. But when trying to play ROS1 bag file, rosbag2 will ask to provide a converter for serializing ROS2 msg rather than using the custom rmw_serialize interface to serialize ROS2 msg.
I just want to play a ROS1 bag file with the custom RMW implementation, using its rmw_serialize interface to serialize ROS2 msg.

MichaelOrlov, 2025-05-07:
- Closing as a stale issue as wouldn't be done. Please feel free to reopen this issue if you think it still makes sense.

closed 2025-05-07 by MichaelOrlov