ros2/rosbag2 #329: rosbag2_transport crashes due to unsafe threading

zmichaels11, 2020-03-24:
## Description
Original issue: https://github.com/ros2/rmw_cyclonedds/issues/118
It looks like the E2E tests for `rosbag2_transport` crash due to unsafe usage of `std::async`.

### Expected Behavior
`colcon test --packages-select rosbag2_transport` should pass when using `rmw_cyclonedds`

### Actual Behavior
** clear and concise description of what _actually_ happened. include copied logs whenever possible **

## To Reproduce
** Steps to reproduce the behavior, e.g.
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. Logs print error '...' **

## System (please complete the following information)
 - OS: [e.g. Ubuntu Bionic]
 - ROS 2 Distro: [e.g. Dashing]
 - Version: [release, branch, or commit hash]

## Additional context
** Add any other context about the problem here **

rotu, 2020-03-24:
Can't edit this ticket, so here's some details to add:

To reproduce:
1. `colcon test --packages-select rosbag2_transport --retest-until-fail 10` (the retest-until-fail is because this test can succeed, depending on how lucky you are)
2. `colcon test-result` 

Actual Behavior:

The test case "RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded" sometimes fails.

sometimes the output looks like this:
```
colcon test-result      
/opt/ros/master/build/rosbag2_transport/Testing/20200324-2148/Test.xml: 15 tests, 0 errors, 1 failure, 0 skipped
- test_record
  <<< failure message
    -- run_test.py: invoking following command in '/opt/ros/master/build/rosbag2_transport':
     - /opt/ros/master/build/rosbag2_transport/test_record --gtest_output=xml:/opt/ros/master/clion/test_results/rosbag2_transport/test_record.gtest.xml
    Running main() from gmock_main.cc
    [==========] Running 1 test from 1 test case.
    [----------] Global test environment set-up.
    [----------] 1 test from RecordIntegrationTestFixture
    [ RUN      ] RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded
    
    >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
    This error state is being overwritten:
    
      'array_list is already initialized, at /opt/ros/master/src/ros2/rcutils/src/array_list.c:61, at /opt/ros/master/src/ros2/rcl/rcl/src/rcl/logging_rosout.c:188'
    
    with this new error message:
    
      'Failed to add publisher to map., at /opt/ros/master/src/ros2/rcl/rcl/src/rcl/logging_rosout.c:190'
    
    rcutils_reset_error() should be called after error handling to avoid this.
    <<<
    [INFO] [1585086552.363023089] [rosbag2_transport]: Listening for topics...
    unknown file: Failure
    C++ exception with description "failed to initialize rcl node: Failed to add publisher to map., at /opt/ros/master/src/ros2/rcl/rcl/src/rcl/logging_rosout.c:190" thrown in the test body.
  >>>

Summary: 15 tests, 0 errors, 1 failure, 0 skipped
```

Sometimes the output looks like this:
```
colcon test-result
/opt/ros/master/build/rosbag2_transport/Testing/20200324-2153/Test.xml: 15 tests, 0 errors, 1 failure, 0 skipped
- test_record
  <<< failure message
    -- run_test.py: invoking following command in '/opt/ros/master/build/rosbag2_transport':
     - /opt/ros/master/build/rosbag2_transport/test_record --gtest_output=xml:/opt/ros/master/clion/test_results/rosbag2_transport/test_record.gtest.xml
    Running main() from gmock_main.cc
    [==========] Running 1 test from 1 test case.
    [----------] Global test environment set-up.
    [----------] 1 test from RecordIntegrationTestFixture
    [ RUN      ] RecordIntegrationTestFixture.published_messages_from_multiple_topics_are_recorded
    -- run_test.py: return code -11
    -- run_test.py: generate result file '/opt/ros/master/clion/test_results/rosbag2_transport/test_record.gtest.xml' with failed test
    -- run_test.py: verify result file '/opt/ros/master/clion/test_results/rosbag2_transport/test_record.gtest.xml'
  >>>

Summary: 15 tests, 0 errors, 1 failure, 0 skipped
```

System:
OS: Ubuntu Focal
ROS2 Distro: Foxy
Version: Master

This can also be readily observed in CycloneDDS Nightly Linux CI: http://build.ros2.org/view/Fci/job/Fci__nightly-cyclonedds_ubuntu_focal_amd64/

zmichaels11, 2020-03-24:
Thanks @rotu, I updated the ticket with your information.

Karsten1987, 2020-03-31:
I've just spent a bit of time to see what's happening.

The problem seems to be the creation of the logging publisher. One way to fix the problem is by disabling the `rosout` logging publisher in the `publisher_manager.hpp`:
```
    auto publisher_node = std::make_shared<rclcpp::Node>(
      node_name,
      rclcpp::NodeOptions().start_parameter_event_publisher(false).enable_rosout(false));
```

I can't reliably reproduce this, but I could imagine that the nodes created for subscription conflict with the creation of the publisher nodes. I'll have to dig further in this to see what's the implication of this in terms of code changes. It reminds me a lot about this one though, also motivated by a cyclonedds test failure: https://github.com/ros2/rosbag2/pull/180

rotu, 2020-03-31:
@Karsten1987, yup, it's the logging publisher as can be seen in the stack traces here https://github.com/ros2/rmw_cyclonedds/issues/118 where @nburek steered me to the answer.

Even if you manage to make the code less crashy by disabling the rosout publisher, I don't think it's a good solution. This function is not threadsafe, period. I think we need to either make it safe or use it in a ways that respects its dangerousness.

Karsten1987, 2020-03-31:
if the actual implementation of the hash map is the culprit here, then I am still a bit puzzled where that race condition should occur. We create a dedicated node per publisher (that's an issue in itself), which shouldn't share any map information. Do they?
Is the conflict coming from the publisher and the rosout publisher on the same node? Either way, I think we should dig into this a bit more to actually isolate the problem here. I think this might not be a problem exclusive to rosbags.

nburek, 2020-03-31:
> if the actual implementation of the hash map is the culprit here,
The hash map does not appear to be the underlying culprit. From the stack trace I was looking at the problem is more likely that the hash map is not being used safely in a multithreaded context. The hash map is not written to be thread safe and should not be used as if it is. 

> We create a dedicated node per publisher (that's an issue in itself), which shouldn't share any map information. Do they?
The hash map in question stores mappings from a Node name to the rosout topic for the node. This is done because loggers in ROS only pass around the name of the logger for the node and no sort of handle to the node or rosout topic. 

If you look at the documentation for nodes in rcl, it explicitly states that rcl_node_init is not thread safe (see: https://github.com/ros2/rcl/blob/master/rcl/include/rcl/node.h#L115). I have not gone deep into the rclcpp implementation recently, but I suspect that this also means the Node class constructor is not thread safe and you need to make sure you are only initializing one node at a time. 

> Either way, I think we should dig into this a bit more to actually isolate the problem here. I think this might not be a problem exclusive to rosbags.
Agreed that more investigation should be done to confirm that it is a threading issue, but this is what is currently implied from the stack traces posted on ttps://github.com/ros2/rmw_cyclonedds/issues/118 (see comment: https://github.com/ros2/rmw_cyclonedds/issues/118#issuecomment-603473151). Yes, there is a good chance this problem is not exclusive to rosbags. Documentation may need to be added for rclcpp/rclpy to note the threading issue or else they should be made thread safe.

rotu, 2020-03-31:
I don't think the hash map is the problem here and it's not a problem exclusive to `rosbag`s. It just so happened that `rosbag` created a *very* reproducible example of an otherwise slippery bug. The crash happens when two processes [mutate](https://github.com/ros2/rcl/blob/40ba55bc1743b480b3e1c76f6c3c009e13b07412/rcl/src/rcl/logging_rosout.c#L188) the same [`static rcutils_hash_map_t __logger_map`](https://github.com/ros2/rcl/blob/40ba55bc1743b480b3e1c76f6c3c009e13b07412/rcl/src/rcl/logging_rosout.c#L83) which maps all logger names to nodes across a process.

The culprit does not seem to be the hash map implementation per se, which makes no thread safety claims. You could say that `rcl_node_init` ought to be threadsafe, (which is [documented as *not* the case](https://github.com/ros2/rcl/blob/5fb711cd0bf813a258e21e97e6d663e3be2ef1d6/rcl/include/rcl/node.h#L115)) or that `rclcpp::Node` ought to be ([it doesn't say one way or the other](https://github.com/ros2/rclcpp/blob/99286978f92c30fe171313bf0785d6b6272c3257/rclcpp/include/rclcpp/node.hpp#L94), but it's one hell of a gotcha in an innocuous-looking constructor).

That `rcl_node_init` is "not threadsafe" is actually an understatement - not only is it unsafe to create two nodes at the same time, but also it is unsafe to call it at the same time as `rcl_node_fini` in the same program, since both objects manipulate the hash set.

Karsten1987, 2020-03-31:
Thanks for the insights. This all makes a lot more sense now given that static logger map :)

clalancette, 2020-03-31:
> The culprit does not seem to be the hash map implementation per se, which makes no thread safety claims. You could say that `rcl_node_init` ought to be threadsafe, (which is [documented as _not_ the case](https://github.com/ros2/rcl/blob/5fb711cd0bf813a258e21e97e6d663e3be2ef1d6/rcl/include/rcl/node.h#L115)) or that `rclcpp::Node` ought to be ([it doesn't say one way or the other](https://github.com/ros2/rclcpp/blob/99286978f92c30fe171313bf0785d6b6272c3257/rclcpp/include/rclcpp/node.hpp#L94), but it's one hell of a gotcha in an innocuous-looking constructor/destructor).

This isn't really the right place for this, but my 2 cents.  I think making the `rclcpp::Node` constructor thread-safe is the right thing to do.  We've done that in some other places (like when using rcl_timers), and I think we need to expand that practice.

PR #338 (2020-03-31): fix cyclone tests
fixes #329 
as discussed in #329 this change disables the `rosout` publishers on the publisher test nodes.

also, fyi @emersonknapp this also disables the QoS test as it also fails for cyclone.
closed 2020-03-31 by emersonknapp