ros2/rosbag2 #342: Expose reading selective topic to command line playback

mabelzhang, 2020-03-31:
## Description

Expose the functionality in #302, reading a selective set of topics, to the command line as options under `ros2 bag play`.

The specific command format is up to suggestion. I did not find such a playback flag in the [ROS 1 bag tutorial](http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data), but there was such a format for recording:
```
rosbag record -O subset /turtle1/cmd_vel /turtle1/pose
```
where `-O` specifies the output file, and the remaining arguments are the topics to record.

Maybe playback can use `-I` for the input file? That flag is free right now.

## Related Issues

Depends on #302 

## Completion Criteria

User is able to specify a subset of topics on the command line to play back.
If no topics are specified, all topics should be played back as normal.

## Implementation Notes / Suggestions

Command line args should be added to `ros2bag/ros2bag/verb/play.py`.
Topic list should be added to `rosbag2_transport/include/rosbag2_transport/play_options.hpp`.
Functionality should be added to `rosbag2_transport/src/rosbag2_transport/player.cpp`.

## Testing Notes / Suggestions

End-to-end tests should be added to `rosbag2_tests/test/rosbag2_tests/test_rosbag2_play_end_to_end.cpp`.
See `rosbag2_storage_default_plugins/test/rosbag2_storage_default_plugins/sqlite/test_sqlite_storage.cpp` for existing tests for the filtering functionality.

emersonknapp, 2020-04-24:
Probably need to do a test robustness pass before closing this ticket

mabelzhang, 2020-05-14:
Test stability followup #396, #408
Test disabled in #410
(The feature is tested in multiple packages - rosbag2_cpp, rosbag2_transport, rosbag2_tests. The one in rosbag2_tests is flaky.)

emersonknapp, 2021-07-20:
This functionality has been implemented

closed 2021-07-20 by emersonknapp